raspberrypi/noobs #432: Support https protocol

digitalLumberjack, 2017-08-17:
Hi!

The https protocol is not supported by noobs since the date is never set, and certificate validation fail.

I would like to know what would be the best solution between: 
* **disable certificate validation** when using wget for tarbals (seems too easy and too insecure)
* **add ntp package to buildroot**, waiting for ntpd to set the date once the network is available (problem behind firewalls ?)
* **using the date http header** from a http HEAD request on http://downloads.raspberrypi.org/os_list_v3.json : `Date:Thu, 17 Aug 2017 12:36:15 GMT`  to set the date (too hacky ?)

Have a good day.

maxnet, 2017-08-17:
https do is supported.
But not for fetching the repository list, as we use the date received from that webserver.

digitalLumberjack, 2017-08-17:
Okay I didn't check this :) 
So the 3rd solution is already in place.

Thank you very much, I close.

closed 2017-08-17 by digitalLumberjack