raspberrypi/noobs #433: Support query params in tarball urls

digitalLumberjack, 2017-08-17:
Hi!

Tarballs urls may contain http query params, but an url like https://mydistribution.com/root.tar.xz?source=noobs cannot be processed as MultiImageWriteThread::untar check the file extension without taking in account query params.

lurch, 2018-04-26:
How likely is it for a tarball URL to end with a query param? :confused:

XECDesign, 2018-04-27:
@lurch, it's not so much that it's likely, but that it would be nice to give maintainers some flexibility. For example, they may want to track where their downloads are coming from.

The reason this hasn't been done is that it wouldn't work with older versions of NOOBS. We'd need an os_list_v4.json or an additional query parameter in os.json.

lurch, 2018-04-27:
Given how NOOBS-specific these downloads are, it's hard to imagine them coming from anywhere _other_ then NOOBS? :wink: 

Unless a future version of NOOBS itself (automatically) added these query-strings, e.g. http://downloads.raspberrypi.org/raspbian_lite/boot.tar.xz?noobs=v2.8.1&rpi=3B so that maintainers could see which versions of NOOBS and which models of RPi are downloading their distros? (which is obviously a separate discussion)

XECDesign, 2018-04-27:
> Given how NOOBS-specific these downloads are, it's hard to imagine them coming from anywhere other then NOOBS?

OP maintains Recalbox, so I bet there's a reason for asking. Perhaps the same files are used by the built-in updater or as a part of an alternative install method.

digitalLumberjack, 2018-04-27:
Hi, you are right we used same files for noobs install and recalbox internal upgrade process, and we wanted to specify the source with a query param :)