raspberrypi/noobs #436: Cannot find NOOBS drive?

JerryWester, 2017-08-30:
I just got a fresh Raspberry Pi 3 just last week. Found an old microSD card, wiped it clean, unzipped a fresh copy of NOOBS from the official site on it, etc. Upon powering up the Pi, it shows the normal splash, and seems normal until it gets to the recovery screen. After about ten seconds of waiting, an error pops up saying something along the lines of "Cannot find NOOBS drive" (will update with exact wording soon), then kicks me back out to the shell after confirmation. The shell has the normal linux-y folders (bin, usr, local, etc.), though everything is read-only. Not sure if it matters, but I used a Chromebook to wipe it (recovery tool, then "Erase Recovery Media").

XECDesign, 2017-08-31:
I've just tried the recovery tool and it looks like that just removes all of the partition tables. How did you partition it afterwards?

JerryWester, 2017-08-31:
It leaves an option to format the drive afterwards, in the file explorer.

XECDesign, 2017-08-31:
It looks like that formats the whole device as vfat, rather than partitioning the device and formatting a partition.

XECDesign, 2017-08-31:
If you do the following, it seems to do the right thing:

1. Press `ctrl + alt + s`
1. `shell`
1. `sudo su`
1. `umount /dev/yourdevice*`
1. `fdisk /dev/yourdevice`
1.
```
o
n
p

t
0c
w
```

Then format the card again, without using the recovery tool.

JerryWester, 2017-08-31:
The only problem with this is that I can only use the shell with dev mode enabled, and seeing as my chromebook is one supplied by my school, this isn't possible for me. But I will look into using another computer for formatting. Thanks for your assistance

closed 2017-08-31 by JerryWester

XECDesign, 2017-08-31:
You could try formatting the card using another device like a phone or a camera, if you have one handy.

JerryWester, 2017-08-31:
Sadly I have a dinosaur of a phone, which has no SD card slot, but thank you for your suggestion