raspberrypi/noobs #440: Kernel panic - not syncing: VFS: Unable to mount root fs on unknow-block(179,2)

alexreg, 2017-09-12:
I have a Raspberry Pi 1, and just copied the latest image onto my SanDisk SD card using diskutil (Mac OS X 10.12) and the official instructions, but got the above error on boot. What's going on here?

XECDesign, 2017-09-12:
Could you please specify exactly what you've done and which files you used? Could you also try using SD Formatter (https://www.sdcard.org/downloads/formatter_4/eula_mac/) instead of diskutil.

alexreg, 2017-09-12:
The image file is `2017-09-07-raspbian-stretch.img`, and I've tried both versions of the official Mac OS X instructions. Will try SD Formatter now.

XECDesign, 2017-09-13:
Okay, the confusion arises because you've opened the issue on the NOOBS repo, but you're not using NOOBS.

SD Formatter won't help you, in that case. Did you use Etcher or dd manually? If you used dd, what is the exact command used?

alexreg, 2017-09-13:
The exact command is on the page I gave, like I said. I tried both the one with `rdisk` and just `disk`. If you want me to copy and paste, here it is! 😜

```sh
sudo dd bs=1m if=2017-09-07-raspbian-stretch.img of=/dev/rdisk3 conv=sync
```

alexreg, 2017-09-13:
Ah, I just tried burning with Etcher, and it indicates this SD card is too small! I thought a 4 GB card should be enough??

popcornmix, 2017-09-13:
An 8G card is required for NOOBS. Check the "software setup guide" link from:
https://www.raspberrypi.org/downloads/noobs/

alexreg, 2017-09-13:
That's it, cheers.

closed 2017-09-13 by alexreg