raspberrypi/noobs #442: convert noobs lite to noobs full

JamesPi123, 2017-09-25:
if i download noobs light and i already have a raspbian image can i convert it to noobs full?

procount, 2017-09-25:
It depends what you mean by a "Raspbian image"!
1. A .img(.zip) file that you downloaded
2. A .img(.zip) file that you downloaded, installed to your RPi and customised that you want to install on something else
3. The contents of the /raspbian folder on the rpf download server that is specially formatted for noobs installation.
?

JamesPi123, 2017-09-25:
A .img(.zip) file that you downloaded, installed to your RPi and customised that you want to install on something else

procount, 2017-09-26:
You need to create a custom os.
See the following 3 links for instructions and examples.

https://github.com/raspberrypi/noobs/blob/master/README.md#how-to-create-a-custom-os-version

https://github.com/raspberrypi/noobs/wiki/Adding-custom-OS-version-examples

https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN

closed 2017-10-02 by XECDesign