raspberrypi/noobs #444: Gadget mode?

procount, 2017-10-11:
I was just wondering if I could enable gadget mode for the PiZero. Is there anything preventing this? 
Which drivers do I need to add and are they already included in current kernel that is used? Or do I need to bump the kernel to a later version?

XECDesign, 2017-10-11:
Yes, gadget mode works quite well on pi zero, but I am not sure I follow how it relates to NOOBS.

procount, 2017-10-11:
For headless access to NOOBS from your PC, instead of using Ethernet or wifi.

XECDesign, 2017-10-11:
Haven't tried it from NOOBS. I'd start by getting the behaviour you want working on Raspbian and then seeing what's missing in NOOBS. I wouldn't be surprised if you had to bump the kernel or edit the  config to enable a few more modules for this. Also keep in mind that the reboot_part parameter doesn't exist with the newer kernel, so you'd need to get that fixed as well if you haven't already.

procount, 2017-10-11:
Yes, I tried it in Raspbian when it was first enabled.
I'm guessing I could just adapt the guides here ->https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=129653 to add it to NOOBS, as was done for Raspbian?

Sorry, what do you mean by the reboot_part parameter?

XECDesign, 2017-10-11:
A bit simpler, without messing with dts files: https://gist.github.com/gbaman/50b6cca61dd1c3f88f41

Regarding reboot_part, this no longer works:
https://github.com/raspberrypi/noobs/blob/master/recovery/util.cpp#L143
It would need to pass the reboot partition all the way back here:
https://github.com/raspberrypi/noobs/blob/master/recovery/main.cpp#L56
Since it's not using glibc, you can't just pass the partition number as a parameter:
http://man7.org/linux/man-pages/man2/reboot.2.html
https://github.com/raspberrypi/linux/issues/2095

procount, 2017-10-11:
I assume you mean the gbaman link is for configuring it *after* I have all the drivers installed?

Ah yes, the autoboot.txt problem!
Hopefully I can stick with my current kernel to keep that working.....

> Regarding reboot_part, this no longer works:
> https://github.com/raspberrypi/noobs/blob/master/recovery/util.cpp#L143
> It would need to pass the reboot partition all the way back here:
> https://github.com/raspberrypi/noobs/blob/master/recovery/main.cpp#L56
> Since it's not using glibc, you can't just pass the partition number as a parameter:
> http://man7.org/linux/man-pages/man2/reboot.2.html
> raspberrypi/linux#2095

So how are we supposed to pass the partition number to ::reboot()? rewrite the ::reboot wrapper around the system call?

XECDesign, 2017-10-11:
> I assume you mean the gbaman link is for configuring it after I have all the drivers installed?

Yup

> Ah yes, the autoboot.txt problem!
> Hopefully I can stick with my current kernel to keep that working.....

No, this has nothing to do with autoboot.txt. It's just a change in the kernel inherited from switching to an upstream reset controller driver, if I recall correctly.

> Regarding reboot_part, this no longer works:
>     https://github.com/raspberrypi/noobs/blob/master/recovery/util.cpp#L143
>     It would need to pass the reboot partition all the way back here: [...]

I think so. I've only looked into it briefly and don't remember what the solution I came up with at the time was. I'd probably replace the reboot with a syscall rather than add a wrapper.

procount, 2017-10-11:
Oh, OK.
For now, the drivers all seem to be in buildroot, so I might just give gadget mode a try when I have a minute (it's a background activity for me).

[BTW, do you how many register bits are available to select the partition number through a warm boot? I'm thinking of installing all OSes I have on one hard disk (~38, so about 80 partitions to reference!) Is that still possible? What is the limit?

XECDesign, 2017-10-12:
There is at least one other register that could be used for this purpose, so it could be expanded to more partitions. However, it's not something I can spend time on right now and I don't know if there might be other uses for those registers in the future.
There's also a reserved value which tells bootcode.bin that the pi has halted rather than rebooted, so you wouldn't want that to match an actual boot partition.

procount, 2017-10-12:
No need to spend time on this - just querying. I've left my Pi installing 50 OSes, so I'll see if I hit the limit or not!

XECDesign, 2017-10-12:
Just checked. 6 bits, value 63 means it's a halt.

XECDesign, 2017-10-12:
I suppose another way around it is bunching the boot partitions together at the start and having the root partitions afterwards, but it would be difficult to tell how they line up at a glance.

procount, 2017-10-12:
50 is a bit excessive and 28 is a reasonable typical limit anyway :smile:
I suppose installing the boot to SD card and rootfs to USB would also allow 56 OSes to be installed.

XECDesign, 2017-11-07:
Ok to close?

procount, 2017-11-07:
Yes. If I need to discuss Gadget Mode again, I'll re-open as I've not had time to investigate further yet.
On the number of partitions, I put a limit in PINN so that you can't install a boot partition number >62.

closed 2017-11-07 by procount