raspberrypi/noobs #445: os_list_v3.json errors

procount, 2017-10-11:
There are a couple of inconsistencies in the os_list_v3.json file on the download website that need to be corrected because they do not match their corresponding os.json files:

__screenly__
"os_name": "Screenly Open Source Edition (OSE)",
Should be:
"os_name": "Screenly OSE",i

__tlxos__
"os_name": "TLXOS ", (NOTE: Trailing space)
Should be: 
"os_name": "TLXOS",

XECDesign, 2017-10-12:
Thanks. I've updated the screenly one, but think the tlxos one may have been intentional. What problems does this cause?

procount, 2017-10-12:
If you download a NOOBS OS Distro file-set onto the SD card or USB stick for local installation, NOOBS uses the os.json "name" field for reference, and if no "icon" field is specified, it uses the "name".png file as the icon by default.

When it reads the remote file list os_list_v3.json, it uses the "os_name" field for reference. In this file, the "icon" field is normally specified because it needs a full URL.

If these "os_name" and "name" fields, differ, NOOBS will treat them as 2 separate OSes and will not be able to compare the two to see which is newer.

Also, if you download the files and store them under the os_list_v3.json "os_name" format, then the default icon name constructed by default (because it is missing in the os.json file) for the local file will not be found.

So the os.json and os_list_v3.json files MUST match. I don't know why there are so many differing fieldnames between these 2 files - I think it's mostly historical and for backwards compatibility. It would have been better if these could have been harmonised, but we have to live with it at the moment.

Why do you think the trailing space in TLXOS was intentional? It breaks the expeted usage.

XECDesign, 2017-10-12:
Got it, thanks.

TLXOS is a bit of a special case. It modifies the installed_os.json file and the space may be serving a purpose in the way it does it. It has been a while since I've examined TLXOS closely, so I'd have to ask the devs to confirm.

procount, 2017-10-12:
I hadn't noticed that unusual behaviour. I wonder why? I'll have to take a look. It will break PINN's new features unless I can add a workaround.

procount, 2017-10-12:
Screenly is still showing as unchanged, but I guess it is just taking a while for your mirrors to catch up with the change...?

XECDesign, 2017-10-12:
Just gave it a nudge. Should be fine now.

procount, 2017-10-12:
Yep. That's fixed it. Thanks.

procount, 2017-10-17:
Any news yet from the devs on why TLXOS has a trailing space in its name?
Is it part of their licensing, as I can't see where it is changed, or why else it would need to be.

XECDesign, 2017-10-17:
No, I haven't asked, but I've just dug up this snippet from an old email:

> ...trailing space on "TLXOS " (this is not a mistake!)...

procount, 2017-10-17:
If it's so important for them to have a trailing space, can't they add it to the os.json "name" field as well?
I can't imagine why it would need to be different - they are supposed to be the same - one is for remote installation and one is for local installation.
The trailing space itself is not the problem. It's the fact that it is different between the "os_name" and "name" fields.
I hate to make a special case for just 1 out of 56 operating systems, but I suppose if I need to.,,,,

XECDesign, 2017-11-03:
I'll have to close this one, since that's all the information I've got.

closed 2017-11-03 by XECDesign