raspberrypi/noobs #446: Source for Windows IoT install scripts

mjmeans, 2017-10-13:
Where is the source for the Windows IoT install scripts for NOOBS? Specifically the files collected from the Windows IoT section of the  API call to http://downloads.raspberrypi.org/os_list_v3.json?

procount, 2017-10-13:
Windows IoT installation is quite unusual. Most of the installation is done by the post installation script (https://iottools.blob.core.windows.net/win10iotnoobs/partition_setup.sh) after NOOBS has installed the basics. (This file is referenced in the os_list_v3.json file you quoted)

mjmeans, 2017-10-13:
Yes. I see that. but where is the source for the commands that this scripts references. i.e. the commands in scratch.tar.xz?

lurch, 2017-10-13:
That's a question for Microsoft, not a question for the developers of NOOBS...

closed 2017-10-13 by mjmeans