raspberrypi/noobs #447: "Cannot find the drive with the NOOBS file" while booting with NOOBS_v2_4_4

bizi05, 2017-10-15:
I have formatted my SD card and then copied all the files of NOOBS_v2_4_4 package after that when I tried to boot up my Raspberry Pi3, I am getting  "Cannot find the drive with the NOOBS file" error.

Need help.

XECDesign, 2017-10-16:
What size is the card and how was it formatted?

imess, 2017-10-18:
Exactly the same error here.  Mine's 16GB class10 Sandisk microSD formatted as FAT32 using SD Association's formatter 5.0.

imess, 2017-10-18:
Never mind. I got it working with the v2 formatter.

XECDesign, 2017-10-23:
No response

bizi05, 2017-10-24:
Sorry for the delay response as I was travelling. I tried with both 8GB as well as 16gb Class10 Sandisk Micro SD card. Tried formatting the SD card with FAT32 using multiple utility tools such Windows Diskpart, SDFormatter 4.1, tc. But no help.
Now I am feeling its the hardware issue as the same SD card works perfectly fine on different Raspberry Pi3(My Friend's Raspberry Pi).
Thanks,Bizi

XECDesign, 2017-10-24:
The recommended tool is "SD Memory Card Formatter". If using version 4, then you need to check the size adjustment option. Version 5 should do that automatically. Can't vouch for other windows utilities doing the right thing.

I am not sure how it could be a hardware issue, unless there's a power issue.

Would you be able to get in a state when this happens, press ctrl-alt-f2, log in as root:raspberry and post the content of /var/log/debug?

bizi05, 2017-10-25:
Thanks for the valuable feedback. Sure I will try whatever you have suggested and keep you posted.
Thanks,Bizi

closed 2017-11-03 by XECDesign

wolfen351, 2019-01-12:
Ok i also had this issue with a 16Gb disk. I used Gparted to make the FAT32 Partition only 2Gb, and then copied the files to the disk again, this time it booted fine. I think the NOOBS installer has trouble with 16Gb+ FAT32 partitions

maxnet, 2019-01-13:
>I think the NOOBS installer has trouble with 16Gb+ FAT32 partitions

No it has problems with superfloppy style SD cards that do not have any partition table, and just have a FAT file system on mmcblk0

Related: https://github.com/raspberrypi/noobs/issues/527