raspberrypi/noobs #448: Build error, cannot find file "translation_ca.qm"

paolopugay, 2017-10-27:
How was this issue encountered?
1. Clone and download the project files (.zip) - noobs-master.zip (10-26-2017 - 5:33PM +4GMT)
2. Extract to a (new) folder
3. Open a terminal and type/perform the following commands:
a."sudo apt-get install build-essential rsync texinfo libncurses-dev whois unzip bc qt4-linguist-tools"
b."./BUILDME.sh"

NOTES:
A. that translation files (*.ts) are inside the "recovery" folder, and no ALL file are not modified.
B. encountered using VMWARE Workstation 12, having LINUX LITE 3.6 - xenial
C. Might be similar to #429, but I decided to open a new issue (rather than re-open) as the version/date of NOOBS file was not mentioned 
D. see attached screenshot
[image]

XECDesign, 2017-11-01:
No problems using my build environment, even when downloading the .zip (which I suspected would be the problem). Downloading Linux Lite to test in a VM.

XECDesign, 2017-11-01:
[image]

Still fine. Have I missed something?

paolopugay, 2017-11-01:
Hi XECDesign,

Thanks a lot for the assistance. I will try to re-create your exact same procedure in my VM.
Also, my laptop is not that fast, I will try to provide a feedback as soon as I can.

paolopugay, 2017-11-02:
Hi XECDesign,

The same issue(s) was still **encountered**. However, this time I did manage to find a workaround to have an output. The only thing missing is the language translations. 

[image]

Maybe you can just send me all the ***.qm files**?
Anyway, I can still use this in this condition.

Also, here's the history of the build:

1. Error was encountered
[image]

2. Updating the **icons.qrc** (changing **.qm** to **.ts**)
[image]
[image]

3. re-run .**/BUILDME.sh**. In here, it displayed that the ***.qm** files is missing but it can continue
[image]

4. Build completed
[image]

5. Bash history
[image]

XECDesign, 2017-11-02:
Undo your hacky workaround and go back to the failing state.
What do you have in you `buildroot/output/build/recovery-1.0` directory?
Is the translation_ca.qm file in there?
What happens if you `cd` there and run `./updateqm.sh`?

paolopugay, 2017-11-03:
Hi XECDesign,

I already did a successful build without any issues.

[image]

Thanks for pointing out that I need to check the **./updateqm.sh**.

To answer your questions: 

1. By reverting back in using the original **icons.qrc** file, the ***.qm** files were not created in the **buildroot/output/build/recovery-1.0** directory

2. By running the **./updateqm.sh**, errors occurred:

Please see the details in terminal as the file manager was not updated/refreshed

[image]

[image]

By installing the **qt4-dev-tools**, solves the issues

[image]

[image]

Again, thanks a lot for the assistance :)

XECDesign, 2017-11-03:
I don't think you ran `sudo apt-get install build-essential rsync texinfo libncurses-dev whois unzip bc qt4-linguist-tools`. lupdate comes from qt4-linguist-tools and your screenshot shows it being installed (which wouldn't be the case if it was already installed earlier).