raspberrypi/noobs #454: Just a few questions:)

Haris1977, 2017-11-14:
Greetings. Just a few questions about noob os:

1) Can i install my images (e.g Openelec, raspbian, Recalbox) on a usb stick rather than sd card through the initial process? If yes how? (i know i will still need the sd card for booting..)
2) How much often the availability of OSes on your remote download repository is being updated? For example recalbox is being updated in a really frequent basis..
3) Will i be able to update individual os? E.g Libreelec, Recalbox through the standard procedure?

Thanks

JamesH65, 2017-11-14:
You should really be asking these sort of questions on the forums.

https://www.raspberrypi.org/forums/

Haris1977, 2017-11-14:
Roger that;)

Haris1977, 2017-11-14:
Well...It is a rather noobs os related question and not a question about a specific image;) 

So i guess this is the right place to ask?

Haris1977, 2017-11-14:
Ok thanks:)

closed 2017-11-14 by Haris1977

XECDesign, 2017-11-14:
In general, this is a place to report bugs/issues, not for general support queries.

A lot of what you're asking depends on the OS in question.

I'll address the questions briefly anyway:
1. Yes, if the OS supports it. When you plug in a USB device, there's a dropdown box which lets you select it as the destination device.
2. Most of the files are hosted remotely, so recalbox manage their own releases and update things as they see fit.
3. Depends on the OS

Haris1977, 2017-11-14:
@XECDesign thanks. As for my 1st question, I m just asking this cause i know this technique "prevents" (or minimizes) the "sd card corruption" problem. 

I just do not know if this kind of problem (sd-card corruption) still persists in nowdays software..