raspberrypi/noobs #457: works with CM3L?

gitcandies, 2017-11-21:
Downloaded Noobs 2_4_4 & extracted everything & moved all the files/directories to an SD card.
 Plugged the SD card into compute module lite (actually plug the io expander board)...Expected it to come up to install Raspbian, but it gave a selection (ONLY) for LibreELEC, ....no Raspbian to be found. Tried process twice, got the same result each time. Finally plugged the card into RPI3 & it showed selections for BOTH Rapsbian & Libre ???? Tried swapping a few times & same result each time. Rapsbian choice would not show up on the CM3L.

Finally ran the Raspbian install selection using RPI3 (which takes about 6 minutes), then plugged the card into the compute module lite...up pops Raspbian & desktop (good)...what's going on??

procount, 2017-11-21:
Try editing the recovery.cmdline file and adding the 'showall' option.

gitcandies, 2017-11-21:
sorry, what is the showall function? (or where is it described).  I'd be happy to give it a try

procount, 2017-11-21:
> Downloaded Noobs 2_4_4 & extracted everything & moved all the files/directories to an SD card.

One of those files is called `recovery.cmdline`. If you put the SD card back in your PC, you will see it. Load this file into your favourite text editor.

That file consists of one single line that contains some specific command line options to alter how NOOBS works. Just add the word `showall` somewhere on the line. Save it, then put it back inyour CML and boot it.

Explanation: Noobs will restrict the OSes it shows you to those that are compatible with the current model of Pi. It could be that Raspbian does not indicate it is suitable for the CML. If so, the `showall` command will relax this restriction and show you all OSes. Let me know if this works.

procount, 2017-11-21:
@XECDesign  - In fact, having compared LibreELEC and Raspbian, the Compute Module seems to be missing from Raspbian's os.json files, although it is in Raspbian Lite's. I'm sure there is a list of these model names and variations somewhere, but I'm struggling to find it....

gitcandies, 2017-11-21:
The CM3L (with the SD card) should be essentially an rpi  on a smaller form-factor, minus various peripherals, correct?

Unfortunately, editing files doesn't do much for the first-time user....who would know to look in various files.  Noobs is supposed to be fool-proof (but what really is)

procount, 2017-11-21:
The editing I am asking you to do is just to get it working for you and to confirm the problem. I think it is just an oversight since LibreELEC and Raspbian Lite included the correct codes for the CML. If you confirm it works, I would expect the RPF to fix the issue.

procount, 2017-11-21:
I found the list of model **numbers** at https://elinux.org/RPi_HardwareHistory and copied them to my wiki here -> https://github.com/procount/pinn/wiki/Model-Identification. But it doesn't include the model **names** from the DTB. I have tried to extract the model names from the Device tree source files and have documented them on my wiki here -> https://github.com/procount/pinn/wiki/Model-Names , but I don't think this is complete because I think some model names extracted from /proc/.... have "Rev xxx" appended. 

Ideally I would like to merge these 2 tables for completeness. Is there a definitive list somewhere, or can anyone provide the missing information for me to complete and merge these tables? @pelwell , @XECDesign

XECDesign, 2017-11-21:
It was left out the full Desktop image because of the space constraints. However, given than CM3L exists, it would make sense to add it. Unfortunately, CM3 and CM3L have the same revision and model name, so it's either blocking install on both or allowing it on both (which I've just done).

I am not entirely comfortable pasting stuff from the firmware source to give a definitive answer, but it takes the following form:
```
"Raspberry Pi %s Rev %s", type, rev
```
I think there's enough publicly available data to predict all the values.
https://github.com/AndrewFromMelbourne/raspberry_pi_revision

gitcandies, 2017-11-21:
I just tried adding the showall & it give these choices on the CM3L:
Raspbian, LibreELEC_RPi, LibreELEC_RPi2

Selecting Raspbian , gives a warning:
Warning Incompatible operating....The following OSes aren't supported....Raspbian
...but it seems to install & operate correctly!

procount, 2017-11-21:
@gitcandies  -  Yes, the warnings are inevitable but as expected. At least we identified and resolved the issue, although it causes another slight issue that it looks like Raspbian is now also available on the CM3, but it doesn't have enough memory to be installed.

@XECDesign 
>  CM3 and CM3L have the same revision and model name

That's a shame...
> or allowing it on both (which I've just done)

Thanks. Probably the lack of memory in the eMMC will actually prevent it from being installed so it will just be a minor annoyance.

Thanks also for the link, I'll take a look and match it up with the RPi devices I have. 
I appreciate the problem of posting firmware code, but since the model identification has switched from using the revision number to the model name, I think it makes sense to publish a list of what the model string is for each variant so that the appropriate identifiers can be put in the os.json files. I will try and compete my wiki page. Perhaps you could check it for me when it is complete?

procount, 2017-11-21:
Don't forget http://downloads.raspberrypi.org/raspbian/os.json 😉

gitcandies, 2017-11-21:
Raspbian installed perfectly fine on the CM3L...no memory problem observed

why would the CM3 & CM3L have the same identification numbers?

closed 2018-01-30 by XECDesign

lurch, 2018-04-25:
> why would the CM3 & CM3L have the same identification numbers?

Because they're *identical* except for the fact that one has an on-board eMMC connected to the SoC's SD bus, and the other has zero-ohm links from the SoC's SD bus to some pins on the edge-connector (which the microSD slot on the CMIO3 board then connects to).
I guess the RPF simply decided that they were too similar to warrant giving them separate revision numbers?