raspberrypi/noobs #463: Factory reset: Uninstall all OSs

Adarma, 2017-12-06:
Could you add the option to "factory" reset to NOOBS without any OS installed?
Currently if all OSs are deselected, the install option is disabled and there is no uninstall option.
This would be good to wipe data before returning an SD card which was preinstalled with NOOBS.

XECDesign, 2017-12-06:
There are no plans to add new features to NOOBS in the near future. If you require features NOOBS doesn't currently provide, I would recommend taking a look at https://github.com/procount/pinn

closed 2017-12-06 by XECDesign

lurch, 2017-12-07:
ping @procount :)

procount, 2017-12-07:
Thanks @lurch ;)
Actually, this has been on my roadmap for a looong time and I had a working version a long time ago (v1.3/1.4?). But when I reintroduced it as "wipe disk" it didn't work in all cases, so I have withdrawn it again. No doubt I will get it working again in the future sometime.

I assume this does not just mean deleting the installed OSes (easy) but also restoring the SD card to full capacity like SDformatter does (which means a lot more fundamental changes)?

Adarma, 2017-12-07:
Yes, ideally restoring the SD card to full capacity. I just found myself in the situation of needing to return a RPi starter kit which came with NOOBS preinstalled. I wanted to remove the OS and return it as delivered. Of course, adding the feature to PINN will not help if the card comes with NOOBS. Still, it may be useful when selling a RPi where PINN is used. Bit of a convenience feature for when no second machine is available to download, reformat and reinstall.

procount, 2017-12-07:
In that situation, you could always drop into the recovery shell using ctrl-alt-f2 using the login of root/raspberry and delete the partitions using fdisk. You can even reformat using mkfs if you want. You have the option to delete NOOBS if you delete or reformat /dev/mmcblk0p1. No need for a second computer.

lurch, 2017-12-07:
Sounds like @Adarma probably wants all the partitions apart from `/dev/mmcblk0p1` deleted, and all the unused space overwritten with zeroes, so that the data he had on the SD card can't be recovered afterwards?  At least, that'd be my interpretation of a "factory reset".

procount, 2017-12-07:
`dd if=/dev/zero of=/dev/mmcblk0p2` and wait a long while. :smile:
Otherwise I think it is getting out of scope.