raspberrypi/noobs #466: Ntsc option 4 not working on rca, also doesnt contains retropie

edneyhelene, 2017-12-30:
Berry boot is the only one that works on my tv! i think noobs developers should take a look on that and try to fix noobs!

Pressing 4 doesnt works for rca ntsc output on the online installer!

The offline installer does works only once, when it does reboots i get blackscreen again! Please do fix it!

Also noobs lacks os images, i dont see arch, i dont see retropie, gentoo, and others...you should improve it thank you

JamesH65, 2017-12-31:
We have no plans to make any further changes to NOOBS. If you have a technical problem, try posting a question on the forums.

If BerryBoot works better for you, use BerryBoot!

XECDesign, 2018-01-01:
I can add taking a look at NTSC output to the todo list.

The installer is tested before release, so we'd need more information to figure out why it's not working for you. Any steps which would help us reproduce the issue would be helpful. It sounds like something went wrong with the installation, which could be the result of a power supply or SD card problem.

The distributions you list aren't available in NOOBS because they don't provide NOOBS builds for us to include. We can't maintain releases for third parties.