raspberrypi/noobs #467: Error: "Cannot find the drive with NOOBS files"

noodood, 2018-01-03:
So I bought a Raspberry Pi 3 Model B a while ago from Amazon, along with a 128 GB Samsung EVO MicroSDXC card. I had to format it to FAT32 using the guiformat.exe shown on [raspberrypi.org](https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md). I then extracted the NOOBS zipped folder into the SD card. I plugged my Raspberry Pi into a screen and turned it on, only to see a pop-up saying "Cannot find the drive with NOOBS files". Does anyone know what happened?

XECDesign, 2018-01-03:
Just tried a 128GB lexar card and guiformat.exe and didn't have any major issues. Did you safely eject the card after copying the files?

Dirty-Rotten-Imbecile, 2018-01-07:
I have the same problem. I have tried unzipping directly to the freshly formated card in both linux and windows. I have used both 8gb card and32 gb card. The 8gb card is fine for loading just Raspbian using the same method. The 32 gb card works fine for running Kali Linux or Rpi and running raspbian but the noobs install kicks back this error everytime.

marangae, 2018-01-12:
I ran into the same problem when I extracted the NOOBS archive and copied the (extracted) files over th the MicroSD. Interestingly when I just copied the NOOBS archive directly on my MicroSD and extracted it there, afterwards deleting the zip archive again, it worked. 

Also check the hashes of the downloaded file to verify that the archive wasn't corrupt.

noodood, 2018-02-05:
I figured out that the only way to fix this was to get a new card. A different one, I might add.

closed 2018-02-05 by noodood