raspberrypi/noobs #468: Help with Yocto / Mender Partitions for Noobs

ikkysleepy, 2018-01-24:
I am having difficult for Noobs to want to install my files. I think the problem is in the partitions.json  I have created tarballs that align with each partition. I have 4 Partitions:

1. Boot
2. RootFS A
3. RootFS B
3. Data

The image I use to burn my os is broken down as follows:
```
Disk rpi-basic-image-raspberrypi3.sdimg: 2.9 GiB, 3133145088 bytes, 6119424 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x0bbfe85a

Device                              Boot   Start     End Sectors  Size Id Type
rpi-basic-image-raspberrypi3.sdimg1 *      24576  106495   81920   40M  c W95 FAT32 (LBA)
rpi-basic-image-raspberrypi3.sdimg2       106496 2088959 1982464  968M 83 Linux
rpi-basic-image-raspberrypi3.sdimg3      2088960 4071423 1982464  968M 83 Linux
rpi-basic-image-raspberrypi3.sdimg4      4071424 6119423 2048000 1000M 83 Linux
```

The tarballs are as follows;
```
-rw-rw-r-- 1 sysadmin sysadmin   12M Jan 24 11:27 boot.tar.xz
-rw-rw-r-- 1 sysadmin sysadmin   323 Jan 24 11:29 data.tar.xz
-rw-rw-r-- 1 sysadmin sysadmin  153M Jan 24 11:28 root2.tar.xz
-rw-rw-r-- 1 sysadmin sysadmin  153M Jan 24 11:28 root.tar.xz
```

and the partitions.json is:
```
{
    "partitions": [
        {
            "filesystem_type": "FAT",
            "label": "boot",
            "partition_size_nominal": 66,
            "uncompressed_tarball_size": 22,
            "want_maximised": false
        },
        {
            "filesystem_type": "ext4",
            "label": "root",
            "mkfs_options": ""
            "partition_size_nominal": 1437,
            "uncompressed_tarball_size": 1037,
            "want_maximised": false
        }
		{
            "filesystem_type": "ext4",
            "label": "root2",
            "mkfs_options": ""
            "partition_size_nominal": 1437,
            "uncompressed_tarball_size": 1037,
            "want_maximised": false
        }
		{
            "filesystem_type": "ext4",
            "label": "data",
            "mkfs_options": ""
            "partition_size_nominal": 1437,
            "uncompressed_tarball_size": 1037,
            "want_maximised": true
        }
    ]
}

```

Any help will be appreciated. I found this wiki which has an example of multiple partitions but can't figure out what I am doing wrong:
https://github.com/steilerDev/NOOBS4IoT/wiki/OS-List

Thanks,
-- Jorge

maxnet, 2018-01-24:
- json file is not well formed. Make sure it passes jsonlint
- Make sure name of tarballs matches your label. May want to avoid spaces.

ikkysleepy, 2018-01-24:
Thanks! I almost got it working now.  I have to look into yocto now to see what I have to load. Thx.

lurch, 2018-01-25:
With two different root partitions, it looks like you'll have to customise your `partition_setup.sh` quite a bit.

ikkysleepy, 2018-01-25:
Yeah, it's not booting up now, the second root partition is for backup. I see the partitions are created but the system is not booting up.  This is my script now:

```
#!/bin/sh

set -ex

if [ -z "$part1" ] || [ -z "$part2" ] || [ -z "$part3" ] || [ -z "$part4" ]; then
  printf "Error: missing environment variable part1 or part2 or part3 or part4\n" 1>&2
  exit 1
fi

mkdir -p /tmp/1 /tmp/2 /tmp/3 /tmp/4

# 						Type		SDIMG Part 	   ->  NOOBS Part
mount "$part1" /tmp/1 # boot 		/dev/mmcblk0p1 -> /dev/mmcblk0p6
mount "$part2" /tmp/2 # rootfs a	/dev/mmcblk0p2 -> /dev/mmcblk0p7
mount "$part3" /tmp/3 # rootfs b	/dev/mmcblk0p3 -> /dev/mmcblk0p8
mount "$part4" /tmp/4 # data		/dev/mmcblk0p4 -> /dev/mmcblk0p9

# Modify the root partition
sed /tmp/1/cmdline.txt -i -e "s|root=[^ ]*|root=${part4}|"

# Add Root FS A to root
sed /tmp/2/etc/fstab -i -e "s|^.* / |${part2}  / |"
# Results: 
# /dev/mmcblk0p7 /  

# Add Uboot Partition to fstab on (part 2 - root fs a - /dev/mmcblk0p7 )
echo "${part1} /uboot auto defaults,sync 0 0" >> /tmp/2/etc/fstab
# Results: 
# /dev/mmcblk0p6 /uboot auto defaults,sync 0 0 

# Add Data Partition to fstab on (part 2 - root fs a - /dev/mmcblk0p7 )
echo "${part4} /data auto defaults 0 0" >> /tmp/2/etc/fstab
# Results: 
# /dev/mmcblk0p9 /uboot auto defaults,sync 0 0 
  
umount /tmp/1
umount /tmp/2
umount /tmp/3
umount /tmp/4
```

How can I mount the partition to verify the fstab file?

procount, 2018-01-25:
There's your problem - it's using uboot.
UBoot has several ways to mount the boot and root partitions, but most of them involve a script buried inside the uboot image. You'll have to do a lot more work I'm afraid.
You need to decompress / expand the uboot image, find the script, change the partitions, and put it all back together again. I'm not sure  if NOOBS includes all the tools to do this, so you may have to add some more tools, depending on how it is compressed.
I've not had time to crack this particular nut myself yet, So please post your partition_setup script if you manage to solve it.

XECDesign, 2018-03-06:
No activity

closed 2018-03-06 by XECDesign