raspberrypi/noobs #470: Unescaped left brace in regex is illegal here in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/dell/noobs/buildroot/output/host/usr/bin/automake line 3936.

mwoolweaver, 2018-02-12:
hardware/software

Ubuntu 18.04 Bionic Beaver (GNU/Linux 4.13.0-33-generic x86_64)
gcc version 7.2.0 (Ubuntu 7.2.0-18ubuntu2)
bc version (1.06.95-9build2)
build-essential version (12.4ubuntu1)
unzip version (6.0-21ubuntu1)
qt4-linguist-tools (4:4.8.7+dfsg-7ubuntu1)
libncurses5-dev version (6.0+20171125-1ubuntu1)
rsync version (3.1.2-2.1)
whois version (5.3.0)
texinfo version (6.5.0.dfsg.1-2)
----------------------------------------------------------------

tried running `./BUILDME.sh ` and got the following error. 

[image]

JamesH65, 2018-02-12:
Does the same thing happen on Raspbian? We don't generally support Ubuntu as its is not packaged by us.

procount, 2018-02-12:
My guess (only a guess) is that the version of build tools supplied in Ubuntu 18.04 are too new and have introduced some changes that are not yet supported. I build it ok in Ubunutu 14.04 and I think some changes were made for the GCC version (6?) in Ubuntu 16, but you are trailblazing a bit to use Ubuntu 18. I would suggest using v14 or maybe v16 if you can.

procount, 2018-02-12:
@JamesH65  - The instructions in README.md for rebuilding NOOBS suggest to use Ubuntu as the Host OS for cross-compiling NOOBS on a PC. However, it does not suggest a particular version. I have also built NOOBS on a Pi3 running Ubuntu and Raspbian, but it takes a while, so I wouldn't recommend it. I haven't tried building it on a PC using the "Debian with Raspberry LXDE desktop (or whatever it is called these days)" but I guess that could be an option if the tool versions are compatible with the project files.

JamesH65, 2018-02-12:
Ah, apologies, hadn't realised this was a NOOBS build thing, thought it was a Ubuntu on Pi thing. Ignore my comment above.

Leobaillard, 2018-02-20:
I have the same problem on Debian Buster.

mwoolweaver, 2018-02-21:
could you provide a list of tools && versions used to compile?

Leobaillard, 2018-02-21:
Sure.
- Debian Buster (GNU/Linux 4.14.0-3-amd64)
- gcc 7.3.0 (Debian 7.3.0-3)
- bc 1.07.1
- unzip 6.0-21
- qt4-linguist-tools 4.8.7+dfsg-11
- libncurses5-dev 6.0+20171125-1
- automake 1.15.1

The source is from the GIT master.

Do you need anything else?

Leobaillard, 2018-02-21:
I just found a patch online which fixes this issue. This is a temporary fix, on an outputed file no less, and it would have to be implemented in the relevant source, but here it is anyway: https://patchwork.ozlabs.org/patch/735705/

franzgl, 2018-03-26:
As written by @Leobaillard you can use his patch.  If you want to fix this with your editor (vi) you can go to line 3936 and make the following changes
$ vi buildroot/output/host/usr/bin/automake
goto line :3936 and change 
  $text =~ s/\${([^ \t=:+{}]+)}/substitute_ac_subst_variables_worker ($1)/ge;
to
  $text =~ s/\$\{([^ \t=:+\{\}]+)\}/substitute_ac_subst_variables_worker ($1)/ge;
----------
The error is with automake and perl v5.26.
In perl v5.22, using a literal { in a regular expression was deprecated, and will emit a warning if it isn't escaped: \{. In v5.26, this won't just warn, it'll cause a syntax error.
---------
@JamesH65 
Please change the NOOBS “How to Rebuild NOOBS” section.
There is only a description for UBUNTU and no description for Raspbian.

XECDesign, 2018-03-27:
Does https://github.com/raspberrypi/noobs/pull/484 fix it?

dlabun, 2018-03-28:
The patch in #484 worked for me.

u2n, 2018-06-18:
@franzgl fix worked for me, allowed to compile pixelserv-tls to run under Optware-ng. Many thanks.