raspberrypi/noobs #473: rpi-wifi-firmware package issues

procount, 2018-02-15:
@XECDesign : There are some issues with the rpi-wifi-firmware package (especially for new developers cloning the noobs github and expecting it to build).

1. rpi-wifi-firmware.mk uses 'master' as the version number, so it will always pull down the latest version rather than a specific version. **(Fix: use a specific commit)**

2. The repo this points to has been upstreamed and the folders have been reorganised. This means it is no longer compatible with rpi-wifi-firmware.mk and it no longer "builds" properly. **(fix: update rpi-wifi-firmware.mk to the new folder structure)**

3. brcmfmac43430-sdio.txt now exists as a patch (https://github.com/RPi-Distro/firmware-nonfree/commit/8d4f3825be09e556c01543119ed50515bd7dd661#comments), so it does not get included by default in the build **(fix: ???)**

So I can resolve points 1&2 as indicated above, but how to fix point 3 so that NOOBS builds "out of the box" again? It is possible to copy the patch into buildroot/package/rpi-wifi-firmware and add it to the NOOBS github. This works, but it won't get updated with the Rpi-Distro repo version.

Is it possible to automate the copying or updating of this patch somehow? It seems a bit of a catch-22 scenario to me: The patch isn't present until the rpi-wifi-firmware is downloaded/patched/configured/built etc by buildroot, but it needs to patch the package before it is configured & built. Somehow the copy needs to take place between downloading the package and patching the package to avoid 2 passes of the build. Any ideas?

maxnet, 2018-02-15:
While not a NOOBS specific problem, the whole `brcmfmac43430-sdio.txt` thing also makes it totally impossible for independent Linux distributions -that are not affiliated with any specific board- to create an universal Linux distribution that runs on more than one ARM board that has this wifi chip.

Perhaps the foundation could exercise some pressure on the wifi chip vendor to dump the .txt file and get the information moved to device tree?

XECDesign, 2018-02-15:
I had no idea the repo was referenced by NOOBS. Will address it when I get a chance. Thanks for the heads up.

XECDesign, 2018-02-15:
Haven't had a chance to test it, but https://github.com/raspberrypi/noobs/pull/474

procount, 2018-02-15:
:cool: That was quick. Looks good. 
I'll be testing it later once I've update it in PINN. :smile:

PR #474 (2018-02-15): Update rpi-wifi-firmware
Fixes #473
closed 2018-02-15 by XECDesign