raspberrypi/noobs #475: Failed to download buildroot source archive when rebuilding

Leobaillard, 2018-02-21:
Hi,

I'm trying to build a custom version of NOOBS and the `BUILDME.sh` script stops after downloading a linux source from GIT and then trying to fetch from `http://sources.buildroot.net` an archive that does not exist, resulting in a 404 and the build script aborting.

This is what the output looks like:
```
>>> linux 85e4adfba269d01050024d03614f30ebcdd4c897 Downloading
Doing full clone
Clonage dans le dépôt nu 'linux-85e4adfba269d01050024d03614f30ebcdd4c897'
Looking up github.com ... done.
Connecting to github.com (port 9418) ... 192.30.253.113 done.
remote: Counting objects: 6200468, done.
remote: Compressing objects: 100% (67/67), done.
remote: Total 6200468 (delta 57), reused 36 (delta 36), pack-reused 6200365
Réception d'objets: 100% (6200468/6200468), 1.73 GiB | 971.00 KiB/s, fait.
Résolution des deltas: 100% (5168966/5168966), fait.
fatal: not a tree object
--2018-02-21 11:22:18--  http://sources.buildroot.net/linux-85e4adfba269d01050024d03614f30ebcdd4c897.tar.gz
Résolution de sources.buildroot.net (sources.buildroot.net)… 104.27.166.48, 104.27.167.48, 2400:cb00:2048:1::681b:a630, ...
Connexion à sources.buildroot.net (sources.buildroot.net)|104.27.166.48|:80… connecté.
requête HTTP transmise, en attente de la réponse… 404 Not Found
2018-02-21 11:22:18 erreur 404 : Not Found.

package/pkg-generic.mk:73 : la recette pour la cible « /home/leobaillard/git/tvl/noobs/buildroot/output/build/linux-85e4adfba269d01050024d03614f30ebcdd4c897/.stamp_downloaded » a échouée
make: *** [/home/leobaillard/git/tvl/noobs/buildroot/output/build/linux-85e4adfba269d01050024d03614f30ebcdd4c897/.stamp_downloaded] Erreur 1
```
(sorry for the French parts)

- Debian Buster (GNU/Linux 4.14.0-3-amd64)
- gcc 7.3.0 (Debian 7.3.0-3)
- bc 1.07.1
- unzip 6.0-21
- qt4-linguist-tools 4.8.7+dfsg-11
- libncurses5-dev 6.0+20171125-1
- automake 1.15.1

procount, 2018-02-21:
I got that error too.
Just download it from https://github.com/raspberrypi/linux/archive/85e4adfba269d01050024d03614f30ebcdd4c897.tar.gz
and put it in buildroot/dl

I would be interested to see if you get this 14.4 kernel version to work as I get a runtime error with: 
```
[   1.013413] bcm2835_thermal soc:thermal: invalid resource
[   1.013562] bcm2835_thermal soc:thermal: Could not get registers: -22
[   6.691962] usb 1-1: device descriptor read/64, error -110
```

Leobaillard, 2018-02-21:
Nice thanks! I'll try that and tell you if it worked.

lurch, 2018-02-21:
`fatal: not a tree object` seems to be the significant part, i.e. the "git clone / checkout process" failed, and that's why it then did the fallback option of trying to download from buildroot.net (which obviously fails, because buildroot.net doesn't mirror the RPi's kernel sources).
So I wonder why the `git` part failed?

procount, 2018-02-21:
Maybe that commit is broken because if you go to here -> https://github.com/raspberrypi/linux/tree/85e4adfba269d01050024d03614f30ebcdd4c897 then it errors with "Failed to load latest commit information"

XECDesign, 2018-02-21:
Loads okay here. Sounds like a temporary github issue.

procount, 2018-02-21:
It also seems to happen on a lot of other branches, not just 4.14.y, so maybe it's unrelated, or more widespread. 🤷‍♂️ Who knows?

lurch, 2018-02-22:
Hmmm, I just did a full local clone of https://github.com/raspberrypi/linux and even though I can see  https://github.com/raspberrypi/linux/commit/85e4adfba269d01050024d03614f30ebcdd4c897 online, when I try to do `git checkout 85e4adfba269d01050024d03614f30ebcdd4c897` in my local clone it tells me `fatal: reference is not a tree: 85e4adfba269d01050024d03614f30ebcdd4c897`, even though other checking out other commits e.g. `git checkout 5739f6bb7f03d1d0c34333b8f893d1ee3747afda` works fine.
So maybe `85e4adfba269d01050024d03614f30ebcdd4c897` has somehow got 'unlinked' from the git repo's history, which is why that commit doesn't exist in local clones? :man_shrugging:

procount, 2018-02-22:
What about 9411792647f688d1401dab486ed01ac4d190a4f4 - will that checkout locally?

lurch, 2018-02-22:
> What about 9411792647f688d1401dab486ed01ac4d190a4f4 - will that checkout locally?

Yep!
```
$ git checkout 9411792647f688d1401dab486ed01ac4d190a4f4
Checking out files: 100% (12762/12762), done.
Note: checking out '9411792647f688d1401dab486ed01ac4d190a4f4'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 9411792... SQUASH: mmc: Apply ERASE_BROKEN quirks correctly
```

procount, 2018-02-22:
That's interesting, because it also exhibits the "browse error" on github. 
So one could conclude (with a sample size of 2?) that the browse error is more extensive than the underlying github problem, which is encouraging.

I think someone should raise an issue nevertheless.

lurch, 2018-02-22:
Maybe this is relevant too https://superuser.com/questions/333997/how-to-pull-a-git-repo-from-github-with-all-unlinked-commits

> That's interesting, because it also exhibits the "browse error" on github.

Yeah, I suspect the "browse error" is probably just a generic github website timeout error, and might just occur for all commits above a certain size or something?

> I think someone should raise an issue nevertheless.

Okay, will do ;-)

procount, 2018-02-22:
What's the difference between the following commits?
85e4adfba269d01050024d03614f30ebcdd4c897 &  068ec703d7b8b920c18c16e9eb8120b7ab91a0e0 ?

lurch, 2018-02-22:
Looks like the [latter](https://github.com/raspberrypi/linux/commit/068ec703d7b8b920c18c16e9eb8120b7ab91a0e0) is actually the same as the [former](https://github.com/raspberrypi/linux/commit/85e4adfba269d01050024d03614f30ebcdd4c897), with the difference being that the latter _is_ actually on the `rpi-4.14.y` branch. And so a `git checkout 068ec703d7b8b920c18c16e9eb8120b7ab91a0e0` on a local clone actually works :-)

Do you want to PR the fix @procount or shall I ?

procount, 2018-02-22:
Is it a fix?
It looks like it to me, but I will try building it later this evening and maybe do a comparison as well.
If it checks out ok, at least it will prevent the git clone error. :sunglasses:

procount, 2018-02-22:
This wouldn't build for me.

make[1]: Leaving directory `/home/procount/noobs_test/buildroot/output/build/linux-068ec703d7b8b920c18c16e9eb8120b7ab91a0e0'
>>> linux 068ec703d7b8b920c18c16e9eb8120b7ab91a0e0 Installing to target
# Install modules and remove symbolic links pointing to build
# directories, not relevant on the target
# Installing dtc (device tree compiler) as host tool, if selected
if grep -q "CONFIG_DTC=y" /home/procount/noobs_test/buildroot/output/build/linux-068ec703d7b8b920c18c16e9eb8120b7ab91a0e0/.config; then /usr/bin/install -D -m 0755 /home/procount/noobs_test/buildroot/output/build/linux-068ec703d7b8b920c18c16e9eb8120b7ab91a0e0/scripts/dtc/dtc /home/procount/noobs_test/buildroot/output/host/usr/bin/dtc ; fi
>>> linux 068ec703d7b8b920c18c16e9eb8120b7ab91a0e0 Installing to images directory
cp /home/procount/noobs_test/buildroot/output/build/linux-068ec703d7b8b920c18c16e9eb8120b7ab91a0e0/arch/arm/boot/zImage /home/procount/noobs_test/buildroot/output/images
# dtbs moved from arch/<ARCH>/boot to arch/<ARCH>/boot/dts since 3.8-rc1
cp /home/procount/noobs_test/buildroot/output/build/linux-068ec703d7b8b920c18c16e9eb8120b7ab91a0e0/arch/arm/boot/dts/bcm27*.dtb /home/procount/noobs_test/buildroot/output/images/
cp: cannot stat ‘/home/procount/noobs_test/buildroot/output/build/linux-068ec703d7b8b920c18c16e9eb8120b7ab91a0e0/arch/arm/boot/dts/bcm27*.dtb’: No such file or directory
make: *** [/home/procount/noobs_test/buildroot/output/build/linux-068ec703d7b8b920c18c16e9eb8120b7ab91a0e0/.stamp_images_installed] Error 1

lurch, 2018-02-23:
> EDIT: I did a comparison between the 2 commits and there are some differences, but nothing major stands out.

Oh yeah - comparing the two commits in more detail, I see that https://github.com/raspberrypi/linux/blob/85e4adfba269d01050024d03614f30ebcdd4c897/Makefile says it's 4.14.15 and https://github.com/raspberrypi/linux/blob/068ec703d7b8b920c18c16e9eb8120b7ab91a0e0/Makefile says it's 4.14.18
(so I wonder if that means that 85e4adfba269d01050024d03614f30ebcdd4c897 became a "dangling commit" when the `rpi-4.14.y` branch got rebased from 4.14.15 to 4.14.18 ? :man_shrugging: )

Afraid I don't have time to dig into this in more detail ATM, so we'll have to wait to see what @XECDesign or @maxnet say :wink:

procount, 2018-02-23:
Yes, I agree with that, especially as when you switch commits it doesn't seem to recognise it has already downloaded the git repo, so it _could_ have just done a checkout and copy. But it tries to download the whole repo again. I have resorted to downloading the linux-{commit}.tar.gz manually first instead (~155MB).

Looking on the rpi-4.14.y branch, 068ec703d7b8b920c18c16e9eb8120b7ab91a0e0 says it's 4.14.18 and a16134b082346b7e7c34f594a0763eafacdcea92 is 4.14.15.
But both of these have missing bcm27*.dtb files that _are_ present in 85e4adf, so they won't build with the current buildroot linux.mk file. 

Maybe some device tree upstreaming has gone on here? I'm out of my depth, so I'll also wait for a new recommendation from @XECDesign for a 4.14 linux commit. :wink:

procount, 2018-02-23:
Current .config has
```
# BR2_LINUX_KERNEL_CUSTOM_TARBALL is not set
BR2_LINUX_KERNEL_CUSTOM_GIT=y
```
Would it be sufficient to change it to:
```
BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
#BR2_LINUX_KERNEL_CUSTOM_GIT is not set
```
Or something similar? Just thinking aloud...

procount, 2018-02-23:
This seems to work, except the file is called `{commitID}.tar.gz` in buildroot/dl instead of `linux-{commitID}.tar.gz`
```
diff --git a/buildroot/.config b/buildroot/.config
index 80df603..7297e68 100644
--- a/buildroot/.config
+++ b/buildroot/.config
@@ -3 +3 @@
-# Buildroot 2015.02-git-00052-geb9a777-dirty Configuration
+# Buildroot 2015.02-git-00055-gfc12128-dirty Configuration
@@ -305,2 +305,2 @@ BR2_LINUX_KERNEL=y
-# BR2_LINUX_KERNEL_CUSTOM_TARBALL is not set
-BR2_LINUX_KERNEL_CUSTOM_GIT=y
+BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
+# BR2_LINUX_KERNEL_CUSTOM_GIT is not set
@@ -309,3 +309,2 @@ BR2_LINUX_KERNEL_CUSTOM_GIT=y
-BR2_LINUX_KERNEL_CUSTOM_REPO_URL="git://github.com/raspberrypi/linux.git"
-BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="85e4adfba269d01050024d03614f30ebcdd4c897"
-BR2_LINUX_KERNEL_VERSION="85e4adfba269d01050024d03614f30ebcdd4c897"
+BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="https://github.com/raspberrypi/linux/archive/85e4adfba269d01050024d03614f30ebcdd4c897.tar.gz"
+BR2_LINUX_KERNEL_VERSION="custom"
```

XECDesign, 2018-02-23:
Should be fixed by 995960e8836e268a16bc7e791bd166ea730323fd

I don't want to switch to downloading tarballs from github because there have been issues with them from the past (which I seem to have trouble finding a reference for right now).

closed 2018-02-23 by XECDesign

Leobaillard, 2018-02-23:
Thanks for the quick fix! You rock!