raspberrypi/noobs #480: Support for touchscreen

Thosty, 2018-03-09:
Is there any way to support the following touchscreen in noobs? In raspbian it works out of the box.

> [  171.960270] usb 1-1.2: new full-speed USB device number 6 using dwc_otg
> [  172.094391] usb 1-1.2: New USB device found, idVendor=2575, idProduct=0401
> [  172.094410] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [  172.094418] usb 1-1.2: Product: CoolTouchR System
> [  172.094426] usb 1-1.2: Manufacturer: Weida Hi-Tech
> [  172.114981] input: Weida Hi-Tech CoolTouchR System as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:2575:0401.0003/input/input2
> [  172.115824] hid-multitouch 0003:2575:0401.0003: input,hidraw1: USB HID v1.10 Device [Weida Hi-Tech CoolTouchR System] on usb-3f980000.usb-1.2/input0

XECDesign, 2018-03-10:
Not in the official release, but if you would like to support it in your build, you'd just need to enable the appropriate option in the kernel config.

The defconfig files are:
https://github.com/raspberrypi/noobs/blob/master/buildroot/kernelconfig-recovery.armv6
https://github.com/raspberrypi/noobs/blob/master/buildroot/kernelconfig-recovery.armv7

For general information:
https://www.raspberrypi.org/documentation/linux/kernel/building.md
And the kernel's own documentation.

procount, 2018-03-10:
On a similar vein, I just got a Hyperpixel from Pimoroni.
Display works ok, but touchscreen only has a python driver which I guess means there isn't a mainstream kernel driver for it.
So I'm thinking of trying to port it. I've no idea what chip it is (mstar 2133A perhaps?). Any advice before I embark on this rabbit hole? Thought I might base it on ADS7879-I2C driver.....?

XECDesign, 2018-03-10:
Quite often python devs just jump straight into python even when an upstream driver is available, so it's worth double checking.

Are you able to figure out what the controller they're using is?

procount, 2018-03-10:
Not yet. It's hidden under ribbon cable. Don't know if there is a tiny connector to remove it and don't want to risk pulling it.
I posted on their forum to gadgetoid but no response yet.
Most touchscreen drivers just seem to specify the parameters and call a module macro to do the real work which I've not found yet.
2 bytes in i2cdump seem to be constant: 0x11 0x23. Maybe an ID so maybe not 2133 chip?

Thosty, 2018-03-10:
@XECDesign Thank you for your answer. Can you please explain me how i can find out the appropriate options for my touchscreen?

XECDesign, 2018-03-10:
Not off the top of my head. I would recommend asking on the raspberry pi forum.

Thosty, 2018-03-10:
Ok

closed 2018-03-10 by Thosty

maxnet, 2018-03-10:
Try running `lsmod` before and after connecting the touch screen's USB, to find out module used.

procount, 2018-03-13:
Hyperpixel uses a CY8C20466 chip, but it's a PSoC, so probably some proprietary touchscreen code. I don't hold out any hope for any existing Linux kernel driver. 
So, as it is not generic, and specific for the Hyperpixel, I was thinking of combining it with code to initialise the display as well, so I can have one Hyperpixel driver to do both jobs. Will this to be frowned upon?