raspberrypi/noobs #481: 3B+

procount, 2018-03-14:
@XECDesign - Ok, so you know what I'm going to ask, don't you! ;)
What happened to noobs v2.5?
What's the difference between 2.6 and 2.7?
Do I have to update the firmware for the new 3B+ (I guess so...)
What do I need to include for PINN - just new firmware or new kernel as well?
When will you have time to update github?
(My 3B+ is already on order!)

XECDesign, 2018-03-14:
We lost the v2.5 floppies. v2.6 had a firmware snafu. You'll need new kernel and firmware. It will be more obvious when I push everything to github, but I won't be doing that before I hibernate - it has been a busy few days.

procount, 2018-03-14:
Floppies!? 5.25" ones maybe?
I suppose 4.9.80 kernel. I have a development version on 4.14, but I have one wifi dongle not working so I'll probably up/downgrade.

procount, 2018-03-14:
if v2.6 is broken, perhaps you should delete it from https://downloads.raspberrypi.org (noobs & noobs_lite), because anyone downloading it from there directly may choose it instead of v2.7, as it is at the top of the folder.

XECDesign, 2018-03-14:
It's using the latest 'master' firmware and kernel 4.9.80 with some patches (to avoid revealing the 3 B+ before release). There are also kernel config changes, I think and updated wifi firmware for the new chip

XECDesign, 2018-03-15:
Pushed all the commits.
The notable ones for PINN are probably these ones:

Enable support for the new LAN chip
https://github.com/raspberrypi/noobs/commit/f1d9a855099e631714122eef8f5e0ea642f688e9

Bump the kernel version and apply the patches which weren't on github
https://github.com/raspberrypi/noobs/commit/c88bbe3e92815d875b871d28c94144bb40964c65

New wifi firmware
https://github.com/raspberrypi/noobs/commit/0b556b0a25bb30dca762a4c11f1756981e6b3e9c

New pi firmware
https://github.com/raspberrypi/noobs/commit/8ed6448e80fe9455e49b6b74b4eaf186d64c03eb

procount, 2018-03-15:
Cheers! 👍 My 3B+ has apparently arrived at home, so I shall have a fun evening! 😃

procount, 2018-03-16:
Hmmm... having a few problems to get a bootable system.

The new kernel is obviously necessary to get the new LAN and Wifi working, but is it also necessary to get it to actually boot? I mean, If I update the firmware, and use the old 4.1 or 4.4 kernel (with new or old dtbs), should it at least boot without the new peripherals working? It doesn't seem to.

I didn't see any kernel config changes you mentioned apart from enabling the LAN, plus any included in the additional patches, but again they're mostly for the new peripherals.

PINN will boot If I use the NOOBS recovery.img files. Maybe I'll have to go back and check I can build the new 4.9 kernel for NOOBS first. Odd that I could build the kernel for 4.14 for a Pi2/3 but not get a 4.9 kernel working on a pi2 or new 3B+!

procount, 2018-03-16:
Ahh, I think I'm missing some kernel config setting changes from my 4.14 branch, since I'm starting from a 4.4 kernel, not a 4.9 where noobs already was!
(Often, one only has to explain the problem for clarity to ensue)

lurch, 2018-03-16:
> (Often, one only has to explain the problem for clarity to ensue)

Indeed: https://en.wikipedia.org/wiki/Rubber_duck_debugging
:smile:

procount, 2018-03-16:
What should KERNEL_DTS_NAME be?
Acc to linux.mk, it should be $(BR2_LINUX_KERNEL_INTREE_DTS_NAME), but that doesn't seem to exist in .config 
What am I missing?

XECDesign, 2018-03-16:
If I remember how this works correctly, it doesn't matter. That variable ends up being used to set `$(KERNEL_DTBS)`. Instead of manually keeping track of the dtbs, I've set it to build all dtbs relevant to the platform:
https://github.com/raspberrypi/noobs/commit/e385dfaa3d589204d27c3160ad654d48846ee308#diff-03658f2fadbdfb904bf63f8af9ccc6d2L231

procount, 2018-03-16:
Except, I've gone back to using the NOOBS .config, * kernelconfig files, and linux.mk without my mods, and yet it's still asking me to fill-in missing kernel config settings in -silentoldconfig
```
  Build a Device Tree Blob (DTB) (BR2_LINUX_KERNEL_DTS_SUPPORT) [Y/n/?] y
    Device tree source
    > 1. Use a device tree present in the kernel. (BR2_LINUX_KERNEL_USE_INTREE_DTS)
      2. Use a custom device tree file (BR2_LINUX_KERNEL_USE_CUSTOM_DTS)
    choice[1-2]: 1
    Device Tree Source file names (BR2_LINUX_KERNEL_INTREE_DTS_NAME) [] (NEW) 
```
are you saying I can put anything here and it'll be ignored?

XECDesign, 2018-03-16:
I don't remember the specifics and am a bit reluctant to look at those makefiles again. All I can say is that NOOBS builds for me as is.

procount, 2018-03-16:
Yeah, NOOBS builds for me too, it's the merging across I'm not doing right. I think I'm just getting muddled between all these different branches and patches, not helped by my additional patches I have to migrate from 4.4 to 4.9.
Oh well, I might have to start again from scratch. Gonna be a long night.....
Thanks.

procount, 2018-03-16:
I finally got PINN booting on my 3B+!
Now to fix a few anomalies and put all the extra bits back in again.

XECDesign, 2018-04-05:
Okay to close?

closed 2018-04-05 by procount

procount, 2018-04-05:
3B+ related: 
I guess at some point we will need to update the 3B+ drivers for the wifi and ethernet, once all the teething problems have been sorted. Are we getting any closer to that yet?

XECDesign, 2018-04-05:
We have top men working on it right now.

XECDesign, 2018-04-05:
Also, what wifi problems?