raspberrypi/noobs #482: Recalbox unbootable

yath, 2018-03-19:
Hi,

on a Raspberry Pi 3 Model B Rev 1.2 and NOOBS 2.7.0, installing Recalbox does not result in a working system. Rather, the Pi switches off after the recovery splash screen (presumably when the OS would boot instead). That same issue happens with Win10, too.

Installian Raspbian or lakka however, works. I even got Recalbox working by replacing the start.elf file on partition 6 with the one from Raspbian. Installing Recalbox (or Windows) directly onto the SD card (via the official download links) also works fine. Is this maybe some incompatibility when chainloading?

Thanks,
Sebastian

XECDesign, 2018-03-19:
I think recalbox issues should be reported here: https://gitlab.com/recalbox/recalbox/issues

yath, 2018-03-19:
Sorry, I don’t think this is a Recalbox issue but one in NOOBS. It happens with other installations targets as well, I just didn’t try all.

lurch, 2018-03-20:
I guess it's up to the Recalbox developers to update their
```
                "https://archive.recalbox.com/v1/noobs/rpi3/boot.tar.xz",
                "https://archive.recalbox.com/v1/noobs/rpi3/root.tar.xz"
```
files to versions which work correctly.

@XECDesign & @procount could this be related to the change in the reset-handling code in newer firmware versions? (which I recall seeing you discussing before). If so, have developers of all the different distros linked to from https://downloads.raspberrypi.org/os_list_v3.json been advised that they need to update the firmware files in their NOOBS tarballs?

XECDesign, 2018-03-20:
The reset handling shouldn't affect the distros themselves. It just sounds like start.elf is not up to date, which is exactly what I'd expect given how recently the 3 B+ was released.

lurch, 2018-03-20:
@yath said he was using a 3B not a 3B+ ?

XECDesign, 2018-03-20:
Yeah sorry, that escaped my mind when I was replying.

procount, 2018-03-20:
> could this be related to the change in the reset-handling code in newer firmware versions? (which I recall seeing you discussing before EDIT: in #390 ). 

I haven't a clue, but from what I recall xecdesign saying previously, it sounds like a mismatch between the bootcode.bin in recovery with start.elf in the various OSes. It might affect autoboot as well. But I haven't had time to check all my OSes yet, as I'm still buried in wifi drivers!

>If so, have developers of all the different distros linked to from https://downloads.raspberrypi.org/os_list_v3.json been advised that they need to update the firmware files in their NOOBS tarballs?

With regards to my OSes: Xbian is (or has) updated. Gentoo64 will update soon. Diet PI has just updated to v6, but I don't know if this includes 3B+ or not. I don't know about the others. If it is just a start.elf problem, I may be able to slipstream it into my OSes when I rebuild them, like I did previously.

On a similar issue, I note that there has been no attempt to distinguish if an OS is suitable for a 3B+ compared to a 3B, so it relies on each OS being updated quickly, otherwise many will be installed that users think are suitable for a 3B+, whereas they are only suitable for a 3B until they are updated. Would it be possible to change the os_list_v3.json to make the OSes more specific to the 3B only (excluding the 3B+) until the appropriate OSes have been updated?

XECDesign, 2018-03-20:
os_list is updated upon request by distro maintainers. I can't keep track of who supports what when and chase everybody down. There's an argument for erring on the side of caution of disabling pi3 support for all non-raspbian distros until contacted by the maintainer, so I'll go ahead and do that.

You'd need to replace "Pi 3" with "Pi 3 Model B Rev".

procount, 2018-03-20:
Do you mean: "disabling pi3B+ support" ?
I know what you mean about keeping track of them - I think I have over 20 Pi3 OSes to check!

On a brighter note, I think I've ported all my working code back onto my main branch and kept it working, so I should be able to release soon, albeit with a few missing wifi drivers for the time being.

XECDesign, 2018-03-20:
Yes, already edited.

yath, 2018-03-20:
@procount 
> I haven't a clue, but from what I recall xecdesign saying previously, it sounds like a mismatch between the bootcode.bin in recovery with start.elf in the various OSes. It might affect autoboot as well. But I haven't had time to check all my OSes yet, as I'm still buried in wifi drivers!

Indeed, when I replace the start.elf in Recalbox with the one from Raspbian, it boots just fine.

lurch, 2018-03-20:
> I can't keep track of who supports what when and chase everybody down.

IIRC that's part of what `os.json` was originally for - `os.json` was supplied by the OS maintainers and explicitly specified which models of Pi the current version of the OS supported, and `os_list.json` was then automatically updated by aggregating info from all the separate `os.json` files. But I recall you saying elsewhere that that update script is currently broken.
(although even if the update script _did_ still work, the waters are also muddied slightly by some of the OSes' NOOBS tarballs now being hosted externally, which means the OS maintainers can update their tarballs without bothering to submit an updated version of `os.json`)

XECDesign, 2018-03-20:
Yes, the script is broken and since os.json is no longer hosted by us, it wouldn't work anyway. Maybe we need a cronjob to crawl through all the os.json files and update the relevant fields in os_list.

matthuisman, 2018-03-20:
So, does "Pi 3 Model B Rev" mean the B or the B+ or both?
I'm looking at: https://downloads.raspberrypi.org/os_list_v3.json

Raspbian just shows "Pi 3" but pretty sure it has been updated to support the B+?
Or maybe the NOOBS tarballs haven't yet been updated?

Then LibreELEC shows "Pi 3 Model B Rev" but no "Pi 3"?
Does that mean only works on B+?
Or do you not need "Pi 3" if you have "Pi 3 Model B Rev"?

Is there a new hex_revision for the new board?

XECDesign, 2018-03-20:
The full name has a format like "Raspberry Pi <Version> Model <Model> Rev <Revision>" (not exactly, but I'm using it as an example). NOOBS matches a substring.

So "Pi 3" would match all pi 3 models, B and B+. However, "Pi 3 Model B Rev" would not match Pi 3 B+, since that would be "Raspberry Pi 3 Model B Plus Rev 1.3", but it would match Pi 3 B, since that would be "Raspberry Pi 3 Model B Rev 1.2".

Hope that makes sense.

matthuisman, 2018-03-20:
Oh I see!
Thank you!

Was their a new hex version?
Or are these pretty much not used anymore?

You may also want to change LibreELEC_RPi2 to "Pi 3" as their NOOBS version is now 8.2.4 
http://releases.libreelec.tv/noobs/LibreELEC_RPi2/os.json
(which supports Pi3B+: https://libreelec.tv/2018/03/libreelec-krypton-8-2-4-mr/)

XECDesign, 2018-03-20:
`a020d3`
https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md

But I don't think theey're used on newer versions of NOOBS.

ianByrne, 2018-03-21:
@XECDesign I just picked up a 3B+ and am finding that the `Windows 10 IoT Core` OS is not available - I see in the `os_list.json` that it looks like it is not (yet?) compatible. Is this something that Microsoft will need to address, or is there some config file or something that I can hack to get the OS installed?

ianByrne, 2018-03-21:
It looks like their hosted `os.json` might be updated (it contains "Pi 3"):
https://iottools.blob.core.windows.net/win10iotnoobs/os.json

```JSON
{
  "name": "Windows 10 IoT Core",
  "icon": "/mnt/os/Windows_IoT/Windows_IoT.png",
  "release_date": "2016-08-02",
  "description": "Windows for devices. Enables you to build projects and apps.",
  "marketing_info": "/mnt/os/Windows_IoT/marketing.tar",
  "url": "http://windowsondevices.com",
  "bootable": true,
  "supported_models": ["Pi 2", "Pi 3"]
}
```

matthuisman, 2018-03-21:
@ianByrne 

"Pi 3" is what nearly every OS has been using for a long time to target the original Pi3.
So I doubt they have actually updated it.
It's just that everyone used "Pi 3" instead of "Pi 3 Model B Rev".
Obviously as this problem wasn't foreseen (maybe need to be more explicit going forward?).

It's unfortunate that "Pi 3" catches both models which makes most of the os.json's currently "in-correct".
If it was called "Pi 4" for example, there wouldn't be the same issue.

NOOBS / PINN doesn't use the os.json to check compatibility.
It uses the os_list_v3.json file (https://downloads.raspberrypi.org/os_list_v3.json)

As you can see, most OS's have all had "Pi 3" changed to "Pi 3 Model B Rev". 
This is to disable them showing on the Pi3B+ until they have been tested / updated.

Raspbian still has "Pi 3" as it has been updated to work on both.

Hope that makes sense?

ianByrne, 2018-03-21:
Thanks @matthuisman , that does make a lot more sense than MS actually being on the ball and updating something like this so soon.

So does that mean that raspberrypi.org needs to update the `os_list_v3.json`, presumably after it has been tested and confirmed works on the new 3B+?

In the meantime, I have manually created the files onto the SD (based off the paths in `os_list_v3.json`), and it appeared in the OS list and seems to install. Although it's not quite booting yet :)

lurch, 2018-03-21:
@XECDesign So, you're now using "Pi 3 Model B Rev" to match the 3B only, and "Pi 3" to match both 3B and 3B+.
I wonder if things might be a bit clearer if you instead used "Pi 3 Model B Rev", "Pi 3 Model B Plus Rev" to indicate support for both the 3B and 3B+ ? (As @matthuisman says, perhaps it's better to be explicit?)
Presumably the OSes will go through the same period of temporary-incompatibility (due to old `start.elf` files) if a theoretical "Pi 3 Model A Plus Rev" got released?

XECDesign, 2018-03-21:
Honestly, I don't want to spend more time editing json files than I absolutely have to.

yath, 2018-03-21:
I see, thanks. So Microsoft (and everyone else) has to rebuild their OS image to work with the new bootloader? Do they know about that?

ianByrne, 2018-03-21:
Looks like Microsoft [will need to do a bit more](https://social.msdn.microsoft.com/Forums/en-US/547a6084-7c7f-40da-911a-37bc6e087d6f/raspberry-pi-3b?forum=WindowsIoT) than just rebuild their OS image for it to work on the new 3B+:

> Raspberry Pi 3B+ is not currently enabled with Windows 10 IoT Core. There are some underlying changes(see below) in the device. The team is working on enablement, but no ETA.

XECDesign, 2018-03-21:
> The issue is that NOOBS 2.7 uses a different technique to boot into an OS than NOOBS 2.4, so it has had some knock-on effects for some distros that haven't got an up to date firmware (bootcode.bin in NOOBS must be compatible with start.elf of each distro). So replacing start.elf from the latest Raspbian should fix them temporarily, but it is up to the distro maintainers to update their OSes accordingly.

I don't think that's the problem, but would have to spend a few minutes to confirm.

procount, 2018-03-21:
My best guess without insider knowledge ;) The os and pi hasn't changed - only noobs firmware and the os reboot syscall.

yath, 2018-03-22:
@ianByrne, I am running a *3B*. Not a +.

yath, 2018-03-22:
@XECDesign Is there any way that I could confirm this? Is there a way to get debug output from NOOBS somehow to UART?

XECDesign, 2018-03-22:
@procount there are a few issue being conflated here. The reset driver change had nothing to do with the firmware. It was just a switch from our own downstream driver within the kernel, to the upstream one, which didn't have the 'reboot_part' parameter.

There was another older issue which prevented autoboot.txt from working. The reset mechanism and selection of which partition to boot are related, so there's a possibility there's a possible link. A part of the fix was to save write a bit to let start.elf know which register to read a value from. That ensured that old bootcode.bin worked with newer start.elf, but not vice versa. If Recalbox uses a start.elf from about a year ago or older, then that's possibly the problem.

@yath Yes, but I can't give instructions off the top of my head. What I usually do is press ctrl-alt-f2, log in as root:raspberry and investigate the situation from there. You could `mount -o remount,rw /mnt`, save everything you need to /mnt (like /var/log/debug), `sync` `mount -o remount,ro /mnt` or just `umount` it.

I believe you're not running `3 B+`, but if you wanted to check, you could look at `/proc/device-tree/model`.

I aim to check whether I can reproduce the issue you're seeing today, but really, it should be reported to the maintainers of the distros in question. Raspbian work, so there's no fundamental problem with NOOBS. It's not outside the realm of possibility that a change to NOOBS has resulted in those distros not working anymore, but it would be up to them to update their distros to be compatible.

Still if NOOBS did break compatability, it would be good to understand what happened and how it can be avoided in the future.

yath, 2018-03-22:
> What I usually do is press ctrl-alt-f2, log in as root:raspberry and investigate the situation from there. You could mount -o remount,rw /mnt, save everything you need to /mnt (like /var/log/debug), sync mount -o remount,ro /mnt or just umount it.

That won’t help me, since the installer/recovery boots just fine. The system halts after (or when) attempting to load the actual OS, at which point I can no longer copy any logs around.

I’ve tried getting the UART to work, which would allow me to cat /var/log/debug > /dev/ttyAMA0 or something similar, but neither combination of enable_uart=1, dtoverlay=rpi3-disable-bt, dtparam=uart1=on, dtoverlay=uart1,txd1_pin=14,rxd1_pin=15, … (ttyAMA and ttyS0 set as a console= in the cmdline) in config.txt did do anything to GPIO14. I’ve even tried renaming recovery.elf to start.elf (as suggested [here](https://github.com/raspberrypi/noobs/issues/354#issuecomment-244320784)) and adding “initramfs”, “kernel=” and ”cmdline=” settings to config.txt, which made the system actually boot, but no output to the serial port either.

I realize that you don’t care whether a change in NOOBS causes *existing stuff* to break, and I, frankly, don’t do either anymore. I just figured it made a nicer user experience, but it’s obviously up to you how you manage your project.

closed 2018-03-22 by yath

XECDesign, 2018-03-22:
> That won’t help me, since the installer/recovery boots just fine. The system halts after (or when) attempting to load the actual OS, at which point I can no longer copy any logs around.

If it's failing in the way we think it's failing, there won't be any useful output. There's no output before the kernel loads (which, in your case never happens).

> I realize that you don’t care whether a change in NOOBS causes existing stuff to break, and I, frankly, don’t do either anymore.

That's not quite it and sorry that I came across that way. I care, but once the release happens there's no going back. It's unlikely that this can be fixed within NOOBS. It's also that things are quite busy right now, so dropping everything to focus 100% on this isn't an option.

matthuisman, 2018-03-22:
@yath
Could you use an older version of NOOBS (2.6) in the meantime?
Or try using PINN which hasn't yet been updated with the NOOBS 2.7 changes.

XECDesign, 2018-03-22:
*2.4 http://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2017-04-10/NOOBS_lite_v2_4.zip

XECDesign, 2018-03-26:
A bit of an update. The firmware used by recalbox is newer that reset register handling change. Recalbox boots and is accessible over the network, but due to some other firmware incompatibility issue the display doesn't come on.

Recalbox will fix this when their next release comes out.

XECDesign, 2018-03-26:
The issue seems to have been introduced here:
https://github.com/raspberrypi/firmware/commit/23047785b7414111cb7cb80aa9d0042c99fae437

So distros with firmware older than 28th of July will have the problem.

matthuisman, 2018-03-27:
@XECDesign 
Looks like OSMC can be enabled for new Pi3B+ as well.

https://osmc.tv/2018/03/osmcs-march-update-is-here-with-pi-3-b-support/

Versions:
http://ftp.fau.de/osmc/osmc/download/installers/versions_RBP
(2018.03-2 is the latest)

NOOBS os.json
http://ftp.fau.de/osmc/osmc/download/installers/noobs/os-pi2.json
""version": "2018.03-2"

samnazarko, 2018-03-27:
Thanks for the ping @matthuisman.

We published updated NOOBS images on the day but supported_hex_revisions and supported_models were not updated. My understanding is that these parameters are superseded by os_list_v3.json; but that all other parameters (such as release_date, kernel and description) are taken from the JSON defined in os_info. Some clarification here would be appreciated.

Feel free to mark as supported.

matthuisman, 2018-03-27:
@samnazarko Your os.json is correct.
It uses "Pi 3" which matches both the Pi3B and Pi3B+.
You can also be more explicit and have "Pi 3 Model B Rev" & "Pi 3 Model B Plus Rev" if you wish.

os_list_v3.json just needs to be updated back to "Pi 3" to enable it for Pi3B+

As for the hex_revision, you could add "**20d3**" but it seems these arn't even used anymore and even https://downloads.raspberrypi.org/os_list_v3.json hasn't added the new hex code for Raspbian.

lurch, 2018-03-27:
Yeah, I can't remember the exact details but I think it's only older versions of NOOBS which use hex_revision, and those older versions of NOOBS won't even look for `os_list_v3.json` (and nor will they boot on a Pi3B+).

XECDesign, 2018-03-27:
Thanks. OSMC enabled as well.