raspberrypi/noobs #483: Tarballs over secure connection?

Sataur, 2018-03-22:
Hi!
I wonder how most tarballs from os_list_v3.json is obtained over plain HTTP?
For example, it is clear that [rpi.thinlinx.com](https://rpi.thinlinx.com) server's security certificate is revoked!

lurch, 2018-03-22:
I'm not sure what the question is, but see also #290 

> For example, it is clear that rpi.thinlinx.com server's security certificate is revoked!

And yet none of the rpi.thinlinx.com links in https://downloads.raspberrypi.org/os_list_v3.json use `https://`.

Sataur, 2018-03-22:
The question is who can guarantee [this](https://tech.feedyourhead.at/sites/default/files/inline-images/OpenElec-Angriff-Schema_0.png) won't happen?

lurch, 2018-03-22:
If you're on a network where you're worried about DNS hijacking, then I'd suggest that you don't use the online-install features of NOOBS. Notice that https://github.com/raspberrypi/noobs/blob/master/recovery/config.h#L25 is itself only an `http://` URL (for reasons explained in #432 ) which means that (AFAICT) it's only the `downloads.raspberrypi.org` domain that would need to get spoofed. (which is actually a 'feature' I used when testing NOOBS several years ago :wink: )

Sataur, 2018-03-22:
What if default repo server would contain only server name without protocol indication? On _downloadListComplete_ check again with secure connection?

maxnet, 2018-03-22:
If security was desired it would make more sense to have a S/MIME signed distribution list with the hashes of the images in it, than to use HTTPS.
As that also provides protection against attack factors like a compromised webserver.
(and as bonus does not require the use of a Certificate Authority.
I always found it odd that HTTPS places total trust in third-party CAs owned by network providers and the public sector. The same organizations that are in the best position to intercept your network traffic).

Only complicating factor is that NOOBS already starts extracting the images while they download (so before you know the hash).

XECDesign, 2018-10-04:
Closing due to lack of activity

closed 2018-10-04 by XECDesign