raspberrypi/noobs #486: LibreElec & race condition

procount, 2018-04-02:
Any idea what's happened to LibreElec noobs downloads? All the necessary files seem to have gone.

This also seems to have highlighted a race condition within PINN which I think I have identified now. It will probably exist in NOOBS as well. I hope to confirm shortly.

ghollingworth, 2018-04-02:
The libreELEC mirror is down at the moment

Gordon

procount, 2018-04-02:
OK thanks, Gordon, it seems to have been down a while.

The problem it causes in NOOBS/PINN is as follows:
The "initialising" dialog takes a long time to time-out waiting for LibreElec.png to download.
If a user presses Esc to cancel this dialog, they can continue to install an OS.
However, if the LibreElec.png download now times out during the download, the onError() routine tries to hide and delete the progressdialog _qpd, which the user already cancelled. But this _qpd is now being used for the OS installation progress, so that gets deleted instead. Eventually when the OS has been installed, it causes a seg fault because the _qpd no longer exists. 

I guess this is an odd corner case, but it can happen. It just needs the overuse of _qpd to be fixed.