raspberrypi/noobs #488: Multiple versions of the same OS

Nilpo, 2018-04-10:
Can we add the ability to have multiple versions of the same OS. I'd like to boot two versions of Raspbian side-by-side.

lurch, 2018-04-10:
I think to do that you'd need to give each of the OSes a different name, they both couldn't be called "Raspbian". So you'd need to make a "full NOOBS" image with your modified Raspbian versions.

@procount can probably share more details with you as I believe he's adapted many other OSes during his PINN work.

procount, 2018-04-10:
Use flavours?

Nilpo, 2018-04-10:
Honestly, I'd really like to see an updated version of the Western Digital fork. I really like their use of "project spaces".

maxnet, 2018-04-10:
Are you sure they didn't do that the way lurch described?

procount, 2018-04-11:
I've heard of the WD version, but I've not seen it. I might include something similar in the future in PINN.
Yes, as @lurch says, you need to give each OS instance a separate name. The easiest way to do that is to create a Flavours.json file with 4 different flavours of the same OS. This will give them each a distinct name, but they will all share the same OS installation files. You can do it as a full NOOBS image, or put Raspbian with its flavours,json file on a USB stick and install from there (or from internet if you have your own repo).

Nilpo, 2018-04-12:
Thanks for the insights. I'll take a closer look and see if that's what they are doing and if I can replicate it.

lurch, 2018-04-12:
@procount If you're curious, it looks like the WD version(s) of NOOBS can be downloaded from http://wdlabs.wd.com/downloads/

Nilpo, 2018-04-12:
It looks like they just have a separate folder for each in the OS folder. Each one is a duplicate image (in this case Raspbian Lite). Seems simple enough. I should just be able to copy in the newest firmware files, right?

Nilpo, 2018-04-14:
You guys were spot on. Thanks for the point in the right direction. I've actually made a community edition of the software with updated firmware and Raspbian Stretch images.

[https://github.com/PiDrive/CommunityFoundationEdition](https://github.com/PiDrive/CommunityFoundationEdition)

This replicates how they did it. I definitely think that flavours sound like a better option since so many of the OS files are duplicated.

Nilpo, 2018-04-14:
You guys were spot on. Thanks for the point in the right direction. I've actually made a community edition of the software with updated firmware and Raspbian Stretch images.

https://github.com/PiDrive/CommunityFoundationEdition

This replicates how they did it. I definitely think that flavours sound like a better option since so many of the OS files are duplicated.

closed 2018-04-14 by Nilpo

lurch, 2018-06-20:
@procount Does that assume that every OS will always be one FAT partition and one EXT4 partition? What happens if am OS has a FAT partition and 2 EXT4 partitions? :wink: (IIRC LibreELEC used to do something like that? No idea if they still do)

procount, 2018-06-20:
Yes, it does assume that, so it doesn't account for all OSes, probably about 80% of them.
LibreELEC is ok, but some of the others aren't, especially WInIoT and Android plus some others.

But this is not a restriction - It is possible to create other dummy ProjectSpace OSes with different numbers of partitions and sizes as you want, but when you get to the remaining 20%, it might be easier to just create some flavours of the specific OS of your choice and install all of those.

lurch, 2018-06-20:
Does installing an OS onto a "Project Space" check that the OS you're installing has the "correct" number of partitions for the ProjectSpace?

procount, 2018-06-20:
Yes. And also that the existing partitions are large enough, because I'm just re-using them, not re-partitioning. It's not gparted after-all :wink: