raspberrypi/noobs #489: Windows IoT core fails to install with NOOBS v2.7

anphel31, 2018-04-17:
When selecting "Windows IoT Core" with NOOBs 2.7, the progress bar appears stuck at 0% for the duration of the install, and the OS does not boot after the NOOBs process completes. The system boots to a completely black screen.
This is reproducible with multiple SD cards, on both RPI2 and RPI3.
The previous NOOBs release v2.4 is working properly.
Any thoughts on how to investigate the issue?

lurch, 2018-04-18:
Possibly a duplicate of #482 ?

XECDesign, 2018-04-18:
That may explain why it doesn't boot, but not the 0% progress bar.

XECDesign, 2018-04-20:
Ideally, MS should update things on their end. I don't know what issues using the old bootcode.bin with new start.elf may introduce. I see at least some 3 B+ specific code in there, so it's possible the hardware won't be initialised properly and suffer from difficult to diagnose stability issues.

XECDesign, 2018-10-04:
Closing due to lack of activity

closed 2018-10-04 by XECDesign