raspberrypi/noobs #490: Raspbian installed by NOOBS 2.8.0 seems to use kernel 4.9.80

neuralassembly, 2018-04-23:
Raspbian installed by NOOBS 2.8.0 seems to use kernel 4.9.80
although Raspbian 2018-04-18 uses kernel  4.14.34.

lurch, 2018-04-23:
@XECDesign Looks like https://downloads.raspberrypi.org/raspbian/os.json says `"release_date": "2018-04-18"` but the `os/Raspbian/os.json` file in the `NOOBS_v2_8_0.zip` says `"release_date": "2018-03-13"` ?

(and on a side-note, perhaps the `"kernel":` field in https://downloads.raspberrypi.org/raspbian/os.json needs updating too?)

lurch, 2018-04-24:
...and just out of curiosity, how come the "offline NOOBS" v2.8.0 includes Raspbian, LibreELEC_RPi and LibreELEC_RPi2 whereas "offline NOOBS" v2.7.0 only included Raspbian?

XECDesign, 2018-04-24:
Ah, I see what happened. The full NOOBS archive was generated from the wrong Raspbian release. Well spotted, everyone. Will try to get a NOOBS v2.8.1 release out today.

@lurch LibreELEC wasn't included in v2.7.0 because we were just trying to get something with 3 B+ support out ASAP. There was no time to get a version of LibreELEC with 3B+ support in there.

lurch, 2018-04-24:
I was just curious what makes LibreELEC "special", and why that gets included in offline NOOBS when no other 3rd-party OS does?

XECDesign, 2018-04-24:
I believe it's included based on the fact that it's relatively small, hugely popular, the devs are always on top of keeping it up to date and SD card distributors requested it to be included.

XECDesign, 2018-04-24:
Uploaded earlier today.

closed 2018-04-24 by XECDesign

lurch, 2018-04-25:
I just noticed that `NOOBS_v2_8_1.zip` appears to contain two copies of each of the LibreELEC tarballs? i.e.
```
os/LibreELEC_RPi/LibreELEC_RPi_Storage.tar.xz
os/LibreELEC_RPi/LibreELEC_RPi_System.tar.xz
os/LibreELEC_RPi/Storage.tar.xz
os/LibreELEC_RPi/System.tar.xz
```
and
```
os/LibreELEC_RPi2/LibreELEC_RPi2_Storage.tar.xz
os/LibreELEC_RPi2/LibreELEC_RPi2_System.tar.xz
os/LibreELEC_RPi2/Storage.tar.xz
os/LibreELEC_RPi2/System.tar.xz
```
:question: