raspberrypi/noobs #491: Raspbian JSON files

lurch, 2018-04-25:
I've just noticed that http://downloads.raspberrypi.org/raspbian/os.json and http://downloads.raspberrypi.org/raspbian_lite/os.json list `"Pi Compute Module 3"` in their `"supported_models"` list. Does this mean that NOOBS will only be able to install Raspbian on the CM3 and not on the CM1 ?
(I believe the fix would just be to change the string to `"Pi Compute Module"` ?)

Hmmm, digging into the problem a bit deeper is this actually a "bug" in https://github.com/RPi-Distro/pi-gen/blob/master/export-noobs/00-release/files/os.json ?

XECDesign, 2018-04-25:
The short version is that CM1 could only have 4GB of on-board storage so it was decided that Raspbian should not be listed. It's recommended that the image is used instead. You could argue that perhaps it should be added, given that USB boot/root is now an option.

Somewhat related: https://github.com/raspberrypi/noobs/issues/457

lurch, 2018-04-25:
> The short version is that CM1 could only have 4GB of on-board storage so it was decided that Raspbian should not be listed.

Ah-ha. That makes sense, and explains why http://downloads.raspberrypi.org/os_list_v3.json says:
* Raspbian supported on NOOBS for CM3 only (since it might be a CM3L)
* Raspbian Lite supported on NOOBS for CM1 and CM3

So I think the fix is to edit just http://downloads.raspberrypi.org/raspbian_lite/os.json so that it matches what http://downloads.raspberrypi.org/os_list_v3.json says about Raspbian Lite?

XECDesign, 2018-04-25:
pi-gen is a bit dumb and currently can't handle the concept that a different stage export may support different hardware. Since the field from os_list is used instead anyway, it's not a priority.

closed 2018-10-04 by XECDesign