raspberrypi/noobs #493: Duplicate tarballs in NOOBS v2.8.1

lurch, 2018-04-26:
I recently noticed that `NOOBS_v2_8_1.zip` appears to contain two copies of each of the LibreELEC tarballs? i.e.
```
os/LibreELEC_RPi/LibreELEC_RPi_Storage.tar.xz
os/LibreELEC_RPi/LibreELEC_RPi_System.tar.xz
os/LibreELEC_RPi/Storage.tar.xz
os/LibreELEC_RPi/System.tar.xz
```
and
```
os/LibreELEC_RPi2/LibreELEC_RPi2_Storage.tar.xz
os/LibreELEC_RPi2/LibreELEC_RPi2_System.tar.xz
os/LibreELEC_RPi2/Storage.tar.xz
os/LibreELEC_RPi2/System.tar.xz
```

`os/LibreELEC_RPi/partitions.json` and `os/LibreELEC_RPi2/partitions.json` each reference `"System"` and `"Storage"`, and I believe this means that:
```
os/LibreELEC_RPi/LibreELEC_RPi_Storage.tar.xz
os/LibreELEC_RPi/LibreELEC_RPi_System.tar.xz
os/LibreELEC_RPi2/LibreELEC_RPi2_Storage.tar.xz
os/LibreELEC_RPi2/LibreELEC_RPi2_System.tar.xz
```
are all redundant; and thus the `RECOVERY` partition is 234MB bigger than it actually needs to be?

This is more of an "FYI" issue, I don't think saving a couple of hundred megabytes warrants the effort of creating NOOBS v2.8.2 :grinning:

XECDesign, 2018-04-27:
Ah, that's because LibreELEC changed their filenames for proper partition naming, IIRC. They added symlinks on their end so that both versions work, but my script downloaded everything. Thanks for the heads up. Maybe the script should parse os.json and download the files actually referenced there.

lurch, 2018-04-27:
> Maybe the script should parse os.json and download the files actually referenced there.

Funnily enough, I've just been working on a "NOOBS fattener" script to do exactly that, and that's how I spotted this issue... :wink:
I hope to polish it off and upload it over the weekend. The only bits my script misses are the `README.md` and `release_notes.txt` files, as these aren't referenced by any of the online JSON files.

XECDesign, 2018-10-04:
Shouldn't happen anymore.

closed 2018-10-04 by XECDesign