raspberrypi/noobs #494: Strange kernel messages in /tmp/messages (v2.8)

procount, 2018-04-27:
What are these strange kernel messages in /tmp/messages related to the squashfs ramdisk (repeated ^H) ?

```
Jan  1 00:00:01 recovery kern.info kernel: [    1.026788] uart-pl011 3f201000.serial: cts_event_workaround enabled
Jan  1 00:00:01 recovery kern.info kernel: [    1.026853] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
Jan  1 00:00:01 recovery kern.info kernel: [    1.028114] sdhost: log_buf @ 8cd13000 (ccd13000)
Jan  1 00:00:01 recovery kern.info kernel: [    1.102890] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
Jan  1 00:00:01 recovery kern.notice kernel: [    1.103541] RAMDISK: squashfs filesystem found at block 0
Jan  1 00:00:01 recovery kern.notice kernel: [    1.103564] RAMDISK: Loading 27895KiB [1 disk] into ram disk... |^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|
Jan  1 00:00:01 recovery kern.warn kernel: [    1.176851] \^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/
Jan  1 00:00:01 recovery kern.warn kernel: [    1.233440] mmc0: host does not support reading read-only switch, assuming write-enable
Jan  1 00:00:01 recovery kern.warn kernel: [    1.233653] \^H
Jan  1 00:00:01 recovery kern.warn kernel: [    1.233714] Indeed it is in host mode hprt0 = 00021501
Jan  1 00:00:01 recovery kern.warn kernel: [    1.233850] |^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H
Jan  1 00:00:01 recovery kern.info kernel: [    1.253055] mmc0: new high speed SDHC card at address aaaa
Jan  1 00:00:01 recovery kern.warn kernel: [    1.253137] -^H\^H
Jan  1 00:00:01 recovery kern.info kernel: [    1.253574] mmcblk0: mmc0:aaaa ACLCD 29.7 GiB
Jan  1 00:00:01 recovery kern.warn kernel: [    1.253577] |^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-
Jan  1 00:00:01 recovery kern.info kernel: [    1.301160]  mmcblk0: p1 p2 < p5 p6 p7 >
Jan  1 00:00:01 recovery kern.warn kernel: [    1.301167] /^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\
Jan  1 00:00:01 recovery kern.warn kernel: [    1.380128] -^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|
Jan  1 00:00:01 recovery kern.info kernel: [    1.443000] usb 1-1: new high-speed USB device number 2 using dwc_otg
Jan  1 00:00:01 recovery kern.warn kernel: [    1.443075] /^H
Jan  1 00:00:01 recovery kern.warn kernel: [    1.443161] Indeed it is in host mode hprt0 = 00001101
Jan  1 00:00:01 recovery kern.warn kernel: [    1.443277] -^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H|^H/^H-^H\^H
Jan  1 00:00:01 recovery kern.warn kernel: [    1.460160] done.
Jan  1 00:00:01 recovery kern.info kernel: [    1.595883] VFS: Mounted root (squashfs filesystem) readonly on device 1:0.
Jan  1 00:00:01 recovery kern.info kernel: [    1.596522] devtmpfs: mounted
Jan  1 00:00:01 recovery kern.info kernel: [    1.597823] Freeing unused kernel memory: 1024K
Jan  1 00:00:01 recovery kern.info kernel: [    1.683359] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
Jan  1 00:00:01 recovery kern.info kernel: [    1.683385] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Jan  1 00:00:01 recovery kern.info kernel: [    1.684100] hub 1-1:1.0: USB hub found
Jan  1 00:00:01 recovery kern.info kernel: [    1.684196] hub 1-1:1.0: 5 ports detected
```

procount, 2018-04-27:
OK, that makes sense.

But in PINN I'm getting these kernel errors just after the above:

```
Jan  1 00:00:03 recovery kern.info kernel: [    3.366254] VFS: Mounted root (squashfs filesystem) readonly on device 1:0.
Jan  1 00:00:03 recovery kern.info kernel: [    3.366928] devtmpfs: mounted
Jan  1 00:00:03 recovery kern.info kernel: [    3.368239] Freeing unused kernel memory: 1024K
Jan  1 00:00:03 recovery kern.info kernel: [    3.404841] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
Jan  1 00:00:03 recovery kern.info kernel: [    3.404856] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Jan  1 00:00:03 recovery kern.info kernel: [    3.405515] hub 1-1:1.0: USB hub found
Jan  1 00:00:03 recovery kern.info kernel: [    3.405599] hub 1-1:1.0: 5 ports detected
Jan  1 00:00:03 recovery kern.info kernel: [    3.480345] uart-pl011 3f201000.serial: no DMA platform data
Jan  1 00:00:03 recovery kern.err kernel: [    3.516869] print_req_error: I/O error, dev loop1, sector 2
Jan  1 00:00:03 recovery kern.err kernel: [    3.517027] EXT4-fs (loop1): unable to read superblock
Jan  1 00:00:03 recovery kern.err kernel: [    3.517473] print_req_error: I/O error, dev loop1, sector 2
Jan  1 00:00:03 recovery kern.err kernel: [    3.517578] EXT4-fs (loop1): unable to read superblock
Jan  1 00:00:03 recovery kern.err kernel: [    3.517900] print_req_error: I/O error, dev loop1, sector 2
Jan  1 00:00:03 recovery kern.err kernel: [    3.518000] EXT4-fs (loop1): unable to read superblock
Jan  1 00:00:03 recovery kern.err kernel: [    3.518314] print_req_error: I/O error, dev loop1, sector 0
Jan  1 00:00:03 recovery kern.err kernel: [    3.518410] SQUASHFS error: squashfs_read_data failed to read block 0x0
Jan  1 00:00:03 recovery kern.err kernel: [    3.518500] squashfs: SQUASHFS error: unable to read squashfs_super_block
```
Any idea where this is coming from? It doesn;t seem to be from the SD card, but more like a badly fomratted ramdisk 🤷‍♂️

XECDesign, 2018-04-27:
Yeah it looks like there's an issue with your recovery.rfs.. Anything else possibly related in the log (especially around where initrd/initramfs is loaded)?

maxnet, 2018-04-27:
>Any idea where this is coming from? It doesn;t seem to be from the SD card, but more like a badly 
>fomratted ramdisk

So can you mount the thing manually on your normal pc?

```
mount -o loop recovery.rfs /mnt
```

Also make sure you didn't put so much stuff in the file that it is now larger than the 32 MB reserved for it (ramdisk_size in cmdline.txt)...

procount, 2018-04-27:
Ah. No. It's ok. I've worked out what it is.
I've duplicated the find_recovery_drive() function in the init script, but I left some of the checks out, so it's trying to mount invalid drives, that's all. Sorry to trouble you and thanks.

closed 2018-04-27 by procount

procount, 2018-05-09:
@maxnet - if/when I get close to 32 MB for recovery.rfs, can I just increase the size of ramdisk_size in recovery.cmdline, or is there some other limit in recovery.elf or elsewhere?

maxnet, 2018-05-09:
>@maxnet - if/when I get close to 32 MB for recovery.rfs, can I just increase the size of ramdisk_size in 
>recovery.cmdline, or is there some other limit in recovery.elf or elsewhere?

I am not the right person to ask about possible limitations in recovery.elf as I am not involved in firmware development.

But as far as Linux goes: yes, you can increase ramdisk_size.
(And the parameter is normally not necessary when using a more modern initramfs cpio archive, instead of legacy initrd file system image. Reason we are using legacy initrd is that it saves memory though)

procount, 2018-05-09:
Thanks for the info.