raspberrypi/noobs #495: NOOBS self update?

procount, 2018-05-03:
@XECDesign - I'm guessing you're following this [forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=212434&start=50#p1311127) .
What do you think? Is it worth my while to port PINN's self-update feature into NOOBS?
I think the only thing it would affect in your CDN system is to add/publish the BUILD-DATA file (and possibly a README.md file if you want to inform users of the changes).
But there's no point in me starting if you don't want to include this feature.

XECDesign, 2018-05-03:
Last time I floated the idea internally, it didn't take off.

procount, 2018-05-03:
That's a shame. I can't imagine why not. I think it's quite stable now too.
It seems a popular option and it might help to stop these types of threads from popping up.

procount, 2018-05-04:
I did it anyway #496. Do with it as you wish.

XECDesign, 2018-05-10:
Found out a bit more information.

We'd have to guarantee that we can upgrade to any future version from all version of NOOBS which would have this feature. Past experience indicates that this can cause all kinds of unexpected issues. You end up having situations where to upgrade from version A to version Z, you need to upgrade to version C first then K then Z. Of course this is all doable, but it's not something we'd want to commit to at this time. I haven't done self-updating software before, so I'm not familiar with everything that can go wrong, but as a user I have seen a lot of it.

Another issue is the possibility of completely unbootable cards, in case something goes wrong during the update. The prime directive is that we don't write to the recovery partition*. In most cases, when a NOOBS card gets corrupted, you can still reinstall the OS. If there's a problem, while we're writing to the recovery partition, that will no longer be the case.

*Unless we have to.

procount, 2018-05-10:
I appreciate that from a commercial perspective it would require more effort to maintain this if you want to make it perfect, which would pull resources off future developments.
The method proposed is no different to the technique detailed [here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=212452) except it is a lot easier for users to manage. It would be very useful in getting small bugfixes out and support for new models (like the 3B+ or Pi4), thus avoiding all the "my 3B+ won't boot" type issues. 

There have been very few NOOBS updates that would have broken - I think the change in partition layout between v1.4 and v1.5 is the major one, maybe the autoboot problems between incompatible firmware is another. But you could also prevent such breaking upgrades from occurring (if you know about them) by switching the BUILD-DATA source file, a bit like you have switched the os_list-v1/2/3.json files.

Anyway, I was just responding to demand, but I understand your point of view. I'll leave it there in case anyone wants to use it.

closed 2018-05-10 by procount

lurch, 2018-05-15:
> Another issue is the possibility of completely unbootable cards, in case something goes wrong during the update.

Yeah, that was my guess at why there was no desire to implement a self-update feature.