raspberrypi/noobs #498: BUG: "Cannot find the drive with NOOBS files"

procount, 2018-05-09:
See also #467 
I think I figured out why this error is occurring.
IF a user formats their SD card as straight FAT with no MBR, then it is mounted as /dev/mmcblk0.
The check in findRecoveryDrive() to find the recovery partition only looks at partition 1 of each device (i.e. /dev/mmcblk0p1) which does not exist, and therefore the recovery files cannot be found.

The function InitDriveThread::method_resizePartitions() can detect a missing mbr and reformat the SD card appropriately, but requires the user to re-copy the NOOBS files back onto it afterwards. In any case, this is run AFTER findRecoveryDrive().

lurch, 2018-05-16:
@procount If you can figure out how/why some users are managing to format their SD card without an MBR, maybe the instructions could be updated to warn against doing that? :man_shrugging:

procount, 2018-05-16:
I always wondered how users were getting this message. But I discovered the problem when I tried to create an image of PINN for Etcher by loop mounting and got it wrong. I don't quite remember what I did, maybe I forgot to partition it before formatting. Anyway the result was this error message after I had burnt the image to the SD card with Etcher.

procount, 2018-05-16:
In Linux, it is possible to delete all the partitions on the SD card, and then do:
`sudo mkfs.vfat -I /dev/sdX
`
but without the -I option it refuses to do it and tells you so, so it's difficult to do it.

Win32DiskImager always seems to create an MBR.
Windows format will not change the partitions. So if previously formatted with an MBR, this will be kept, but if there is no MBR, it won't add one.

lurch, 2018-05-16:
> Win32DiskImager always seems to create an MBR.

Win32DiskImager doesn't 'create' an MBR, it just copies the image-file onto the SD card verbatim (basically like a Windows version of `dd`). So _if_ the image-file contains an MBR, then that'll get copied to the SD card by Win32DiskImager ;-)

procount, 2018-05-16:
sed "s/Win32DiskImager/SD Card Formatter/g"

procount, 2018-05-16:
A disk without an MBR is also known as a "Superfloppy". Unless it is supplied like that by the manufacturer, it's hard to see how a user can easily create one.

WHK102, 2018-07-29:
Same problem, y try copy files but canot start. The debian with rpd canot boot, all images are corrupted from official website, y use a rpi3+. With centos7 for rpi works fine.

chadberchek, 2018-08-20:
I encountered this problem with a Samsung SDXC EVO 64GB (MB-MP64D). I am not sure if it came this way from the manufacturer because I got it a few years ago and previously used it with a Windows Phone. However I don't recall ever editing partitions on it so it most likely came this way.

I followed the setup instructions. First I tried to format it using the SD Card Formatter tool from sdcard.org. It kept failing, so I did a thorough (not "Quick") format to exFAT with Windows to see if the card was bad. It formatted fine and had no bad clusters. Then I reformatted it to FAT32 using guiformat from Ridgecrop. Then I unzipped NOOBS directly to it and booted to this error message.

I fixed it as follows:
diskpart
select disk #
clean
convert mbr
create partition primary offset=4096
(Where # is the disk number; 1 in my case.) Then reformat to FAT32 using guiformat, copy files again, and boot. 4096 KiB was just a guess for offset; I haven't benchmarked it.

Incidentally, after the original error message, it would drop me into a surprisingly functional Linux shell with lots of programs available, so it seems to read quite a bit from the disk before failing. I don't know anything about the boot process but perhaps it would be possible to make it more intelligent so it can work without an MBR partition.

lurch, 2018-08-21:
Without an MBR partition, you can only have a single 'partition' on the card, which means NOOBS wouldn't be able to create any new partitions to install any OSes onto :wink: 

The "surprisingly functional Linux shell" is all contained in the initramfs (recovery.rfs) and IIRC is a basic buildroot / busybox system, mainly useful for debugging.

Potherca, 2018-09-16:
>  If you can figure out how/why some users are managing to format their SD card without an MBR

I'm getting the "Cannot find the drive with NOOBS files" message for an install done from a Chromebook... Would ya'll be interested in more details?

laneschmidt, 2018-12-14:
Getting this issue myself, anyone find a solution? Here is my [raspberrypi.stackexchange question](https://raspberrypi.stackexchange.com/questions/91937/cannot-find-the-drive-with-noobs-files-when-trying-to-boot-pi-3-b).

laneschmidt, 2018-12-14:
> I encountered this problem with a Samsung SDXC EVO 64GB (MB-MP64D). I am not sure if it came this way from the manufacturer because I got it a few years ago and previously used it with a Windows [...]

What file was this added to? I don't understand fully

vobarian, 2018-12-15:
> What file was this added to? I don't understand fully

The above snipped is not added to a file; they are commands typed at the command line (be sure to launch cmd as admin).

procount, 2018-12-15:
> Getting this issue myself, anyone find a solution? Here is my raspberrypi.stackexchange question .

A solution is not to just format your sd card with gparted, but first ensure you create a partition and then format the partition as FAT32.

Another solution is to use Etcher to write a copy of a Raspbian bersion to your SDCard first (maybe use Raspbian-lite as it is smaller and will therefore be quicker, espwcially as the files will later be deleted). Try booting it to make sure it works. If it does, take the card out and put it in a pc. Now replace all the files on its first boot partition (typically the only partition you will see on Windows) with the contents of the NOOBS zip file. Now return it to your RPi and boot it.

A third solution is to use Etcher to flash the Etcher image version of PINN called: pinn-lite.img.zip directly.

Flashing an image with Etcher writes the MBR as well as the partition contents so it ensures you don't miss out this step.

Agungsetiawanpawallo, 2019-03-11:
> 
> 
> Without an MBR partition, you can only have a single 'partition' on the card, which means NOOBS wouldn't be able to create any new partitions to install any OSes onto 😉
> 
> The "surprisingly functional Linux shell" is all contained in the initramfs (recovery.rfs) and IIRC is a basic buildroot / busybox system, mainly useful for debugging.

How to make MBR partition ?

Agungsetiawanpawallo, 2019-03-11:
> I always wondered how users were getting this message. But I discovered the problem when I tried to create an image of PINN for Etcher by loop mounting and got it wrong. I don't quite remember what [...]

how to burn that with Etcher?

lurch, 2019-03-11:
> how to burn that with Etcher?

https://github.com/procount/pinn/blob/master/README_PINN.md#using-etcher

Agungsetiawanpawallo, 2019-03-12:
> 
> 
> > how to burn that with Etcher?
> 
> https://github.com/procount/pinn/blob/master/README_PINN.md#using-etcher

whats going on if error happend when I burn PINN with Etcher?

lurch, 2019-03-12:
That's hard to answer, if you don't tell us what the exact error was :wink:

joba-1, 2019-08-26:
still an open issue? Isnt everything in here for this to be closed?

OP @procount already found the cause, @lurch pointed out the reason why this is fatal.

Use a better error message, e.g. "Drive with NOOBS files has no partition table" and done. Right?

FWIW, I created my noobs disk like this, after I encountered the error and found the reason here (replace the _ in the first line with the drive letter):

`
d=sd_ \
&& f=~/Downloads/NOOBS_v3_2_0.zip \
&& sudo mkdir -p /dev/shm/$d \
&& sudo dd if=/dev/zero of=/dev/$d bs=1M count=1 \
&& sudo parted /dev/$d mklabel msdos \
&& sudo parted /dev/$d mkpart p fat32 0% 100% \
&& while [ ! -f /dev/${d}1 ]; do sleep 1; done \
&& mkfs.vfat -n 'NOOBS_3_2_0' /dev/${d}1 \
&& mount /dev/${d}1 /dev/shm/${d}1 \
&& cd /dev/shm/${d}1 \
&& unzip $f \
&& cd - \
&& umount /dev/shm/${d}1 \
&& rmdir /dev/shm/${d}1
`
...hope I did not put errors in this while reformatting...

rsmckinney, 2019-09-08:
This is still an issue even with the standard Windows 10 formatting tools.  I had to use the following format tool for the copied image files to be discovered by the pi:

https://www.sdcard.org/downloads/formatter/

Why is this issue not fixed?  I imagine anyone running Windows using their own blank sd card will likely run into this, which represents a pretty sizable chunk of users.  At the very least detect this case and display an instructive message.

Bobbbay, 2019-11-20:
I have been trying to do this on my Chromebook. It's giving me the error. No, I don't have an MBR partition, is there any way to create one ?!? I can format my microSD to fat32, so that's all good... Any ideas? What should I do? 

Thanks, Bobbbay

P.S. I am running Linux (beta) on my Chromebook but the terminal isn't allowed to see my external devices (security reasons, of course. Linux _is_ run in an isolated sandbox, so...).

procount, 2019-11-20:
@Lurch - I'm wondering, if instead of just giving up if we can't find the NOOBS files on Partition1, we give the user the choice of attached drives to select as the boot drive and then let InitDriveThread do its thing (with appropriate warnings etc)? Maybe method_resizepartitions() could call saveBootFiles() as well (if possible), in order to avoid having to copy the NOOBS/PINN files again. What do you think?

lurch, 2019-11-20:
> What do you think?

:man_shrugging: 
I don't work on NOOBS any more, you'll need to ask @XECDesign

lurch, 2019-11-20:
@Bobbbay Maybe you have a friend with a Windows PC who can run SD Formatter for you, to put an MBR onto your SD card?

procount, 2019-11-20:
@Bobbbay Or you could try formatting it in a digital camera 🤷‍♂ ?

XECDesign, 2019-11-20:
I don't really work on NOOBS either, aside from updating it for newer hardware. When I was using a chromebook, I had it in developer mode and was able to use the terminal to get a shell and use fdisk or parted to partition the device, then format the partition. I don't know if things have changed. Maybe there's a better way now.

procount, 2019-11-20:
> I don't work on NOOBS any more, you'll need to ask @XECDesign

@lurch - I know, I know. I just meant it as a general concept, as it still seems to be an issue. 
I'll see what I can do in PINN and if I get it working then maybe @XECDesign  can consider it for NOOBS.
All the code seems to be there to do it, it's just a matter of plumbing it altogether.

procount, 2019-11-20:
The concept seems to work! 🎉 
I need to put some safeguards in though, like:

- Check `runinstaller` flag is present
- Only check for `/dev/mmcblk0` and `/dev/sda`
- Prompt user to confirm drive wipe & format
- Check it works with large OSes present (not just lite version)

Anything else?

lurch, 2019-11-20:
Why only `/dev/sda`? I've never tried it myself, but https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md implies that the Pi should boot from the first MSD on which it finds a `bootcode.bin` file?

procount, 2019-11-20:
Since I'm trying to let the user select from a list, that won't matter anymore (but I just broke it). 😞

lurch, 2019-11-20:
Hmm, why does the user need to select from a list? Isn't the better option to "fix" the code that tries to find the "NOOBS files"? Apologies if I've grabbed the wrong end of the stick.

procount, 2019-11-20:
Maybe you're right. I was conflicted.
https://github.com/raspberrypi/noobs/blob/54ac40f7c59e945d5a1389e356c4531b53ca0216/recovery/main.cpp#L225 seems to indicate that the recovery partition could be on any drive, so findRecoveryDrive() looks at the first partition of all non-virtual drives and selects the first one it finds. 

Is this sufficient? It seems to work because mmcblk0 comes before sdX alphabetically and happens to match the normal boot order. But are there any other device names that could be used apart from mmcblk0 or sdX? What about the compute module? If not, maybe it is sufficient to search just mmcblk0 and sdx devices in that order to follow the normal firmware boot order.

I was thinking about a case where you have a NOOBS formatted USB drive and a NOOBS formatted SD card inserted at the same time. Either one of these could be non-MBR formatted (if it's a first install). So we don't just want to look at the first partitions of the devices, but all partitions of all devices in boot sequence order
. 
So I presume the complete list would be:
mmcblk0, mmcblk0p1..mmcblk0pN, sda, sda1..sdaN, sdb, sdb1..sdbN, ..., sdz, sdz1..sdzN?
(Could anyone confirm this is exactly the same order that the firmware uses for the boot order?)

But when Pi4 USB booting comes online, there may be the possibility to boot from a specific USB drive in preference to the SD card thanks to the BOOT_ORDER setting, which means you have no idea what the real boot drive is. However, I believe there might be the possibility to pass information about the real boot device, so if this becomes available, it may even be easier than the current search method.

So unless I've missed something, it should be possible to fix it without involving the user, although I was being rather cautious to prevent the runinstaller option formatting the wrong drive in case I got it wrong. Comments?

procount, 2019-11-20:
Sometimes the simplest fixes are the best.  Lots of testing ahead, mind :wink:

Bobbbay, 2019-11-20:
So then coding and testing it is! I'm not happy that NOOBS has such a problem - it is meant for NOOBS, right? Kinda sucks that literally putting an OS meant for beginners on an SD or USB is this hard. I never liked that you have to format your external device to fat32, but this too? By the time they read through all the documentation they won't be NOOBS anymore - they'll be complete experts! I mean, either that or they'll just quit. I know that if I wasn't such an experienced a Raspberry Pi developer, I would have done so by now... So may the pull requests begin 🎉!

procount, 2019-11-20:
It's quite a rare problem, but when it strikes it's difficult for a noob to fix it.
I will have a fix in PINN, if you'd like to test it before I issue any pull requests?

Bobbbay, 2019-11-21:
@procount so will this version work without an MBR partition?

lurch, 2019-11-21:
@procount 
> It's quite a rare problem

Indeed, I think that all of the (many!) SD cards I've purchased, have come with "MBR formatting".

> but when it strikes it's difficult for a noob to fix it.

Only in the unlikely event they're unable to run the SD Formatter tool, as suggested in the README? https://github.com/raspberrypi/noobs#how-to-format-an-sd-card-as-fat
:man_shrugging: 

@Bobbbay 
> so will this version work without an MBR partition?

As I outlined in an [earlier comment](https://github.com/raspberrypi/noobs/issues/498#issuecomment-414686033) it's not possible for NOOBS / PINN to work on a card without an MBR, but I believe the fix @procount is suggesting is for PINN to automatically create an MBR on cards that don't have one.

lurch, 2019-11-21:
> It seems to work because mmcblk0 comes before sdX alphabetically and happens to match the normal boot order.

LOL that's fortuitous :)

> But are there any other device names that could be used apart from mmcblk0 or sdX?

I don't think so - AFAIK mmcblk0 will always be the SD-card in the SD-card slot, and any USB MSD will always appear as sdX.

>  What about the compute module?

The eMMC on the ComputeModules still appears as mmcblk0

> I was thinking about a case where you have a NOOBS formatted USB drive and a NOOBS formatted SD card inserted at the same time.

In that case, the Pi would boot from the SD card, so the "look at mmcblk0 before sdX" logic should still apply?

If, OTOH you have nothing in the SD-slot, but two USBs inserted each containing NOOBS, I dunno what the logic is then ;-)  But that's possibly an edge-case it's not worth worrying about?

I'm not sure that you need to check mmcblk0pN and sdaN? I think the bootrom only loads bootcode.bin from the first partition (which also has to be FAT-formatted) ?
And if you have an mmcblk0p1 then I guess that means the SD-card **does** have an MBR?

> But when Pi4 USB booting comes online

Probably not worth worrying about that yet...?

> you have no idea what the real boot drive is

Maybe there's some Linux kernel feature which identifies which device it was loaded from? :man_shrugging:  I have no idea.

> I was being rather cautious to prevent the runinstaller option formatting the wrong drive in case I got it wrong.

I assume you'll only need to reformat if the drive doesn't have an MBR (i.e. mmcblk0 or sda) and you have runinstaller in the kernel cmdline. I guess as a safety check you could also see if the cmdline.txt file on the drive you're about to reformat has runinstaller in it? (if it hasn't already been removed by the NOOBS code by then)

Bobbbay, 2019-11-21:
Well @procount your a genius. The sourceforge worked. I will start tests tomorrow and will have my results by next Thursday. Depending on my results/bug reports, I'll close this thread. Thanks again, Bobbbay.

procount, 2019-11-21:
@lurch - I think I forgot an edge case.
Maybe it should check for bootcode.bin being present as well as recovery.rfs? After all, the firmware uses that to boot.
I think it's highly unlikely to have 2 noobs formatted devices in where the higher priority boot order device doesn't have a bootcode.bin, but it'll make it more robust.

procount, 2019-11-21:
@Bobbbay - I'm not sure you can close my thread, but anyway, that should wait until/if the fix is applied to noobs, not just PINN.

lurch, 2019-11-21:
> Edit: oh, but what about the situation where you only have bootcode.bin on the SD card to boot off a USB device?

If people are enough of an advanced user to do that, they're probably also advanced enough to be able to ensure they're using MBR partitioning? :wink:

procount, 2019-11-21:
This is included in PINN v3.3.2 just released. Let's see how it goes!

Bobbbay, 2019-11-23:
No problems yet, successfully ran Kali and Raspbian Lite. I'm giving a go for pull requests and stable releases! Cheers!

Happy hacking, Bobbbay

P.S. Yes @procount, I can't close your thread, but I think everything here is done!

Bobbbay, 2019-12-01:
This can be closed now, considering there haven't been any objections for a week.

Bobbbay, 2019-12-09:
@procount I think this **should** be closed now. Is the working PINN merged with NOOBS?

XECDesign, 2019-12-09:
I'll leave it open until NOOBS has the same fix.

PR #564 (2020-02-04): Less restrictive search for NOOBS files
Fixes #498 

This removes some restrictions on the drives that NOOBS will search for its files to determine the boot drive.
In this way it will detect NOOBS on a superfloppy drive that has no MBR and will allow it to be partitioned and formatted properly.

I produced some test images to prove it works.

https://sourceforge.net/projects/pinn/files/testing/superfloppy.zip/download is an image file that should be flashed to a blank SD card with Balena Etcher. It contains NOOBS 3.0 on a superfloppy formatted partition. When it boots, it will not detect the N [... cut]
closed 2020-02-04 by XECDesign