raspberrypi/noobs #500: bsdtar

procount, 2018-05-26:
NOOBS uses bsdtar 3.3.1 to unarchive the OS files. But when I try to create an archive with it (using even a basic `bsdtar -cf test.tar .`) It errors with
```
bsdtar: statfs failed
bsdtar: Error exit delayed from previous errors.
```
Any ideas ?

maxnet, 2018-05-27:
Wouldn't count on such exotic functions being fully supported by uclibc.
Might need to switch toolchain to glibc if you want to use that.
(but then I would expect Windows installations to fail, unless they compiled their custom stuff statically instead of against uclibc).

procount, 2018-05-27:
Well, obviously it's for PINN not NOOBS, but I had to check it failed in noobs too to make sure I hadn't messed anything up!

Odd thar bsdtar will decompress but not compress with uclibc. I might have to switch to tar instead which works, or pull it to pieces and fix statfs. Thanks for the pointers.

closed 2018-08-06 by procount