raspberrypi/noobs #501: I'm looking for a tool that determine "mac address" before setup.

backy0175, 2018-06-02:
Hi there, I need your help.

I'm trying to setup Pi Zero WH without monitor & keyboard & mouse (headless). I know how to enable ssh and WiFi through create "ssh" and "wpa_supplicant.conf" with NOOBS. But my network is required "mac address" before connect. (network access is restricted by mac address)

So thought, if NOOBS can execute the recovery command to install OS, it might be possible to execute a tool that gets the Mac address with the same mechanism and saves it on the SD card.

Is there any tool like this? or Is there any good way to determine "mac address" with headless? or How do I create such tool?

Thanks in advance for your help.
backy0175 from Kyoto, Japan

lurch, 2018-06-02:
I can't think of an easy way of doing that if you're totally headless :-/
If you're lucky, @procount might add a feature to do what you're asking for to PINN :wink:

procount, 2018-06-03:
Also being discussed here ->https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=215001

procount, 2018-06-03:
You can do this with PINN version 2.6.2 or later

With PINN on your Sd card instead of NOOBS, add the following 2 files (to the partition where PINN is stored, i.e. /dev/mmcblk0p1, or whatever drive letter it mounts as in Windows)

pinn_init.sh:

```
#!/bin/sh
cp /tmp/mnt/getmac /tmp
/tmp/getmac &

```
getmac:
```
#!/bin/sh
sleep 20
mount -o remount,rw /mnt
ifconfig -a  | grep HWaddr >/mnt/myipinfo
mount -o remount,ro /mnt

```

Put the SD card in your Pi and boot it. If this is the first time you are booting PINN, please wait until the Sd card has been formatted. (Watch the green LED and wait, say, 30 secs after it has been solidly lit for maybe a minute). If PINN has already formatted the card, then you only need to wait about 40 seconds after powering on.

Once the green LED has stopped flashing erratically (or has dropped to maybe 1 regular quick flash every couple of seconds, if you have a USB stick fitted), it is safe to remove the power and return the SD card to your PC. Look for a file called `myipinfo` which will contain the MAC addresses of all network adapters fitted.

procount, 2018-06-04:
Glad to be of help.

As you are running headless, once you have your RPi on your network, it might be worth adding `ssh vncshare forcetrigger remotetimeout=20` to your recovery.cmdline. That way you will be able to control PINN remotely as well. See https://github.com/procount/pinn/blob/master/README_PINN.md for full details, but particularly https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-use-pinn-headlessly-vnc and https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-access-the-shell-or-ssh-into-pinn

XECDesign, 2018-06-04:
Please close the issue if it's resolved.

backy0175, 2018-06-12:
Hi procount,
Thank you for additional information and sorry for reply so late.

In the way your suggestion, I could easily get the MAC address, However, finding the proper WiFi pre config and installing with headless were extremely difficult.

But finally, I successfully installed raspbian with headless and it is now possible to access with SSH.

The difficult point of headless install is that

1)
the WiFi setting depends on the access point authentication method and encryption method, etc.

2)
silentinstall do not copy wifi pre config(wpa_supplicant.conf)

3)
silentinstall with modified partition_setup.sh do not work propery. (I didn't dig into it.)

reference:
"Easy headless Raspbian Wifi Setup (for NOOBS) - Raspberry Pi Forums"
https://www.raspberrypi.org/forums/viewtopic.php?t=141559

But, NOOBS vncinstall with proper WiFi pre config is working fine and done.
I thought that explain the method of headless installation on the official website for beginners, if possible.

Anyway, Thank you for everything.
backy0175 from Kyoto, Japan

closed 2018-06-12 by backy0175

lurch, 2018-06-12:
If you're operating entirely headlessly, I wonder if it'd be user to use 'regular' Raspbian instead of trying to fiddle about to get NOOBS to do what you want? :man_shrugging:

backy0175, 2018-06-13:
Hi lurch,

I would like to use Pi Zero WH for digital circut education for my children. As you know Pi Zero WH is quite compact and cheep. However LCD monitor is expensive more than 5 times of Pi Zero WH. And My children has there own note PC. So headless 'regular' Raspbian is very confortable for install and remote desktop environment through VNC.

lurch, 2018-06-14:
In that case I wonder if you'd find https://www.raspberrypi.org/blog/gpio-expander/ useful? :wink: