raspberrypi/noobs #507: can bootcode.bin be configured to mount a predefined partition and write to it?

d3banjan, 2018-07-27:
Is it possible to make `bootcode.bin` to write a predefined string (i.e recovery partition id) to `autoboot.txt`?

The usecase is that we want the file to signify that start of the bootprocess and also boot into the  recovery (i.e another extended parttion, not noobs recovery) partition automatically if the boot process fails between start to end of boot (signified by start of init.d !).

popcornmix, 2018-07-27:
No. The firmware (bootcode.bin or start.elf) never writes to the sdcard.

d3banjan, 2018-07-27:
What about something a bit later in the boot process?

maxnet, 2018-07-27:
>What about something a bit later in the boot process?

You could modify the NOOBS code to do that.
But that's something you would need to code yourself, rather than there being a standard option for that.

Would set `default_partition_to_boot` parameter in `noobs.conf` on the setting ext4 partition rather than touch `autoboot.txt` or any other file on the FAT partition, though.

d3banjan, 2018-07-27:
thanks @maxnet and @popcornmix ... it is a very common usecase and I was hoping for an easier fix.

d3banjan, 2018-07-30:
@maxnet thanks.. you are right of course - it is not a good idea to write on the FAT partition. 

Since this is a resonably simple operation of writing a predetermined string to a file as early as possible, any idea where in the noobs code can I add this behaviour?

maxnet, 2018-07-30:
You probably want to replace this if block, for the equivalent that writes the hard coded string.

https://github.com/raspberrypi/noobs/blob/master/recovery/bootselectiondialog.cpp#L169

lurch, 2018-07-30:
I wonder if the `before-reboot` feature of [PINN](https://github.com/procount/pinn) (mentioned [here](https://github.com/procount/pinn/issues/174#issuecomment-408011149) ) would do what you want?
Ping @procount

procount, 2018-07-30:
Unfortunately, I removed the `before-reboot.sh` feature in p2.4.4b because it prevented some OSes from booting. See https://github.com/procount/pinn/commit/4b15b329a54c5538263b3f234b32824198d8e92d#diff-81706e5fe8ac70459737d21c007fceddL102 . But feel free to add it in again and debug it for me. :wink:

XECDesign, 2018-07-31:
Since this isn't really NOOBS related, it probably belongs on the forum instead.

closed 2018-07-31 by XECDesign

agl62, 2018-08-03:
Here's the fix! (d3banjan actually asked this on my behalf and I have been trying to find something since)

The default partition to boot (from the NOOBS login screen) can be changed by editing /media/pi/SETTINGS/noobs.conf and changing `default_partition_to_boot` (the settings partition is /dev/mmcblk0p5 in the event that it isn't already mounted to /media/pi/SETTINGS).

From there, you can just do `sudo reboot <partition_number>` and it will ignore `default_partition_to_boot` in `noobs.conf` for only the next reboot, so that on the reboot after it will go back to the default partition :)

lurch, 2018-08-03:
See also #508 , if you're interested in that sort of thing.