raspberrypi/noobs #508: autoboot.txt overrides boot {part}

matthuisman, 2018-07-29:
**NOOBs Lite 2.8**
http://director.downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2018-04-19/NOOBS_lite_v2_8.zip

**Current Behavior**
NOOBS ignores reboot {part} and always boots to boot_partition in autoboot.txt

**Expected Behavhour**
reboot {part} reboots into part and autoboot.txt is ignored.

**Steps to reproduce**

- Format SD card and copy over NOOBs Lite 2.8 files
- Boot NOOBs and install LibreELEC and OSMC 
- Boot LibreELEC, enable SSH
- SSH into LibreELEC (root / libreelec) and do command **reboot 8**
- Observe Pi reboots to OSMC
- SSH into OSMC (osmc / osmc) and do command **sudo reboot 6**
- Observe Pi reboots to LibreELEC

- Put SD card into PC and create autoboot.txt file in root of SD card with below contents
**boot_partition=6**
- Boot Pi
- Observe Pi booting directly into LibreELEC
- SSH into LibreELEC and do command **reboot 8**
- Observe Pi reboots to LibreELEC (and not OSMC as expected)

lurch, 2018-07-31:
AFAIK this behaviour is all entirely in firmware-land and not in NOOBS-land.
Ping @popcornmix

XECDesign, 2018-09-13:
I've had a quick go at fixing this:
[bootcode.bin.zip](https://github.com/raspberrypi/noobs/files/2380027/bootcode.bin.zip)

Seems to work for me, but it would be good to get a few testers to check I haven't messed anything else up. It just needs to be extracted onto the NOOBS partition (the one with recovery.elf) as bootcode.bin.

matthuisman, 2018-09-20:
@thegigibeast
Could you give this a test (above comment)

Dustee, 2018-09-23:
I just tested, and reboot command works as expected.

My setup is : noobs 2.8, rasplex 1.8 (outdated firmware?), libreelec 8.2.5, retropie latest.

XECDesign, 2018-09-24:
Thanks for testing, @Dustee. Seems to work well for me too, except for when booting off USB.

Unfortunately, that requires a change to start.elf. We can fix NOOBS and Raspbian relatively quickly (I hope), but then it will take a while for the change to propagate to the other distros.

matthuisman, 2018-09-25:
@XECDesign

Does they changes break the other distros from booting?
Or, will they simply just continue to have this autoboot issue until they are updated?

XECDesign, 2018-09-25:
The combination of booting off USB, using autoboot.txt AND a reboot parameter would result in files being loaded from two different partitions. If any of the variables are removed, then everything works.

popcornmix, 2018-10-09:
Fix should be in latest rpi-update firmware.

XECDesign, 2018-10-11:
In the process of publishing v2.9 with this fix:
https://github.com/raspberrypi/noobs/releases/tag/v2.9

closed 2018-10-11 by XECDesign