raspberrypi/noobs #509: Boot NOOBS from USB?

IzaacJ, 2018-07-30:
Hello.
My RPi 3B is booting stuff perfectly from USB (RasPlex and others) but I thought I'd show my friend (which is new to RPi) how to use NOOBS, and as far as I understood, booting NOOBS from USB has been implemented?

I downloaded NOOBS_lite_v2_8 and axtracted that to my usual RPi 8GB USB drive and put it in, but I'm stuck on rainbow screen.

Using the power plug and USB cable that I got with my Pi, and have successfully measured that it outputs the needed voltage and amps so I know that it ain't a power issue.

Is there anything I need to do for it to boot? With RasPlex (and other builds) I had to change the commandline file, but it has nothing in it regarding boot/disk, only root. Should I add boot+disk as I have to do with RasPlex?

XECDesign, 2018-07-31:
How long did you wait after the rainbow screen? I've just tried this and it seems to take a very long time for the kernel to load initially, like it's in USB1 mode. Might be an issue with bootcode.bin in the current release.

XECDesign, 2018-07-31:
@ghollingworth , would you happen to know if that's the case?

ghollingworth, 2018-07-31:
It shouldn't...

XECDesign, 2018-08-06:
Closing due to lack to activity.

Since the firmware has to load the squashfs filesystem in addition to the kernel, and it does it block by block, without DMA, it's expected to take a while.

If other distros boot from USB, NOOBS should as well (without any changes).

closed 2018-08-06 by XECDesign