raspberrypi/noobs #510: Accessing Recovery

LePipotron, 2018-08-02:
Hello,
I have a problem with my NOOBs and i'd like to reset it, but i'm at work and my raspberry is at my house, i controll it with VNC and putty. Is there a solution to access recovery mode and re-install NOOBs or maybe a way to totaly reset my NOOBs without touching my raspi.
Thanks

XECDesign, 2018-08-02:
In theory, yes, but I wouldn't recommend trying it. You're very likely to lock yourself out completely.

There's a vnc option, but it doesn't just work, there are additional steps I don't remember off the top of my head. IIRC, if you just enable VNC, it doesn't wait for you to connect, so you also need to use the option that forces it to trigger. There may be others I am forgetting.

You could also get NOOBS to reset itself and then install the distro of your choice, but then you'll also need to make sure you can enable ssh on the guest OS before you boot it.

I don't have an exact step by step set of instructions that's guaranteed to work, but I know it's possible.

procount, 2018-08-02:
What OS(es) have you installed with NOOBS? (Raspbian?)

>  i controll it with VNC and putty.

I guess you mean you can access your RPi through vnc and putty in the installed OS (Raspbian)?
Although NOOBS supports VNC, it is a little known option, and it doesn't support ssh/putty.

> I have a problem with my NOOBs and i'd like to reset it,

What is the problem that requires a reset of NOOBS? And what do you mean by a reset - reinstall NOOBS? wipe out all installed OSes? Change settings? Maybe the solution would be easier if we knew what the problem is.

As Xecdesign states,  it should be possible, but you will have to rely on keeping your installed OS to provide SSH/putty support through the process and one wrong move will lock you out. It would be much easier to do this on PINN since it has SSH support built-in.

XECDesign, 2018-08-02:
> you will have to rely on keeping your installed OS to provide SSH/putty support through the process and one wrong move will lock you out

I was thinking more along the lines of installing a fresh install of Raspbian with the `ssh` file trick rather than keeping the existing setup.

procount, 2018-08-02:
It's not clear to me what the actual problem is: whether they want to keep their installed OS, reinstall NOOBS, wipe the whole SD card, clear the data(?) or indeed if it's a problem with Raspbian that he wants to "reset" (since many newbies refer to a version of Raspbian installed through NOOBS as NOOBS iteself!). Hopefully some more information will be forthcoming soon.....

LePipotron, 2018-08-02:
OOps when i answered i didn't saw all the response.
@procount when i say reset all i mean wipe all and restart to 0 and i can't access physicly to my raspi now

XECDesign, 2018-08-02:
@LePipotron What's the output of 'apt policy'?

XECDesign, 2018-08-02:
Stock sources files:
https://github.com/RPi-Distro/pi-gen/raw/master/stage0/00-configure-apt/files/sources.list
https://github.com/RPi-Distro/pi-gen/raw/master/stage0/00-configure-apt/files/raspi.list

sources.list goes in /etc/apt/
raspi.list goes in /etc/apt/sourced.list.d/

LePipotron, 2018-08-02:
I did it and i had that : 

> Fichiers du paquet :
 100 /var/lib/dpkg/status
     release a=now
 500 http://mirrordirector.raspbian.org/raspbian stable/firmware armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=firmware,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stable/rpi armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=rpi,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stable/non-free armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=non-free,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stable/contrib armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=contrib,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stable/main armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=main,b=armhf
     origin mirrordirector.raspbian.org
 500 http://archive.raspberrypi.org/debian stretch/ui armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=stretch,l=Raspberry Pi Foundation,c=ui,b=armhf
     origin archive.raspberrypi.org
 500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=stretch,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.org
 500 http://www.deb-multimedia.org wheezy/non-free armhf Packages
     release v=7.11,o=Unofficial Multimedia Packages,a=oldoldstable,n=wheezy,l=Unofficial Multimedia Packages,c=non-free,b=armhf
     origin www.deb-multimedia.org
 500 http://www.deb-multimedia.org wheezy/main armhf Packages
     release v=7.11,o=Unofficial Multimedia Packages,a=oldoldstable,n=wheezy,l=Unofficial Multimedia Packages,c=main,b=armhf
     origin www.deb-multimedia.org
 100 http://ftp.debian.org/debian wheezy-backports/non-free armhf Packages
     release o=Debian Backports,a=wheezy-backports,n=wheezy-backports,l=Debian Backports,c=non-free,b=armhf
     origin ftp.debian.org
 100 http://ftp.debian.org/debian wheezy-backports/contrib armhf Packages
     release o=Debian Backports,a=wheezy-backports,n=wheezy-backports,l=Debian Backports,c=contrib,b=armhf
     origin ftp.debian.org
 100 http://ftp.debian.org/debian wheezy-backports/main armhf Packages
     release o=Debian Backports,a=wheezy-backports,n=wheezy-backports,l=Debian Backports,c=main,b=armhf
     origin ftp.debian.org
 500 http://security.debian.org wheezy/updates/main armhf Packages
     release v=7.0,o=Debian,a=oldoldstable,n=wheezy,l=Debian-Security,c=main,b=armhf
     origin security.debian.org
 500 http://ftp.fr.debian.org/debian wheezy/non-free armhf Packages
     release c=non-free,b=armhf
     origin ftp.fr.debian.org
 500 http://ftp.fr.debian.org/debian wheezy/contrib armhf Packages
     release c=contrib,b=armhf
     origin ftp.fr.debian.org
 500 http://ftp.fr.debian.org/debian wheezy/main armhf Packages
     release c=main,b=armhf
     origin ftp.fr.debian.org
Paquets épinglés :
`

XECDesign, 2018-08-02:
get rid of all the debian.org and deb-multimedia entries

LePipotron, 2018-08-02:
Okay thanks for the default sources.list file it worked i did an update and an upgrade and i reboot the raspi and it's working. But now i'm back with my other problem, when i want to install:

> sudo apt-get build-dep python-imaging

it's saying to me that i have to add some adress to the sources.list file...

XECDesign, 2018-08-02:
Check the original source files carefully and read the comments. Remove the `#` that's in front of the `deb-src` lines in both files. Then run `sudo apt update`.

LePipotron, 2018-08-02:
You saved my f day <3

lurch, 2018-08-02:
You shouldn't be using packages from "standard Debian" with Raspbian - IIRC 'armhf' for Debian is ARMv7, but 'armhf' for Raspbian is ARMv6.

XECDesign, 2018-08-02:
With the caveat "unless you know exactly what you're doing". There are cases when you might want to do that, but that's a separate question.

@LePipotron okay to close?

LePipotron, 2018-08-02:
Just one last question, is that normal that i can't install :

> sudo apt-get install libjpeg8 libjpeg62-dev libfreetype6 libfreetype6-dev

it's saying that they are in hold state :/

closed 2018-08-06 by XECDesign