raspberrypi/noobs #512: [os_list_v3.json] Lakka_RPi2 - Please enable for 3B+

matthuisman, 2018-08-13:
http://mirror.lakka.tv/noobs/Lakka_RPi2/os.json
(Version: 2.1.1)

http://www.lakka.tv/articles/2018/08/12/lakka-211-with-raspberry-pi-3-b-plus-support/

https://downloads.raspberrypi.org/os_list_v3.json

needs Lakka_RPi2 supported_models updated to below to enable Pi3B+ support

```
"supported_models": [
    "Pi 2",
    "Pi 3",
    "Pi Compute Module 3"
],
```

kivutar, 2018-08-13:
Oh, good catch.

So which one is the Pi3B+ there?

XECDesign, 2018-08-13:
"Pi 3" should match all pi 3 models, including the B+. I'll update it some time today. Thank you both.

matthuisman, 2018-08-13:
@kivutar

Your os.json is fine :)
As it already uses "Pi 3".

Just the os_list_v3 needs to be updated on NOOBS side.

Over @ PINN, I had added your beta build, so Pi3B+ users could still use Lakka up-till now.
I will remove that once the os_list_v3 is updated.

kivutar, 2018-08-14:
I mailed them.

XECDesign, 2018-08-14:
It should be updated now.

closed 2018-08-14 by matthuisman