raspberrypi/noobs #513: Advice on Host OS (ext4 update)

procount, 2018-08-13:
Currently I use Ubuntu 14.04 to develop on (Linux ubuntu 3.13.0-32-generic #57-Ubuntu SMP), since it is still the recommended OS in the README file.
I avoided upgrading to 16.04 because of the problem with the RiscOS ADFS/Cumana partition tables (#262).
But in trying to convert the latest Kali-linux 2018.2 version to NOOBS format, I found it could not mount or read the ext4 partition. Dmesg indicated: "EXT4-fs (sdb2): error loading journal" & "JBD2: Unrecognised features on journal". 
I managed to mount the partition in Raspbian Stretch, though (Linux raspberrypi 4.14.50-v7+ #1122 SMP Tue Jun 19 ) so obviously Ext4 has some new features that are not compatible with my aging 3.13 kernel of Ubunutu 14.04. (It's not clear to me yet if Kali uses 4.9 or 4.15 kernel)

So does anyone know:
1) What changes have been made to the ext4 journal to make it incompatible (or which kernel version)?
2) Would Ubuntu 18.04 be compatible with the new ext4 journal?
3) Do the later Ubuntu's still have ADFS/Cumana built-in?
4) Should I change to a new host OS? If so, which - Debian with Raspbian Desktop perhaps! ;)

What host OS does everyone else use to develop on?

lurch, 2018-08-14:
> What host OS does everyone else use to develop on?

You seem to be under the impression that other people are also trying to get Kali working with NOOBS...? :wink: :rofl: 

> Do the later Ubuntu's still have ADFS/Cumana built-in?

No idea, but https://github.com/raspberrypi/noobs/issues/262#issuecomment-106082208 shows you how to check.

procount, 2018-08-14:
> You seem to be under the impression that other people are also trying to get Kali working with NOOBS...? 

No, that's just _**my** catalyst_ for thinking about changing my host OS. I realise that won't affect everyone else.
Likewise, the ADFS/Cumana problem in Ubuntu 16.04 doesn't necessarily affect the building of NOOBS, just the mounting of an already NOOBS-formatted SD card/USB stick.
IIRC, there were also initial issues with the version of GCC in Ubuntu 18.04 for building NOOBS, but I think they were resolved.

So before I start having to install a myriad of different OS versions, I was just asking what other people were using as a starting point for candidate OSes to try. The more I think about it, "Debian with Raspberry Pi Desktop" might be a strong candidate as I think it resolves points 1&2, leaving just GCC to check.

maxnet, 2018-08-14:
Think you can just drop the journal with the right tune2fs option and mount it afterwards like it's an ext3 file system...

Am using Kubuntu myself.

procount, 2018-08-14:
@maxnet - Thanks for the hint. 
I did a dumpe2fs and have highlighted what I think are the problematic new features:

**Filesystem features:**      has_journal ext_attr resize_inode dir_index filetype extent **64bit** flex_bg sparse_super large_file huge_file dir_nlink extra_isize **metadata_csum**
**Checksum:**                 **0xb1f2be57**
**Journal features**:         journal_incompat_revoke **journal_64bit FEATURE_I4**

These features seem to come from Stretch (or equivalent). Probably the checksum. But I can't remove it in Ubuntu 14.04 because my e2fsprogs does not support it:

> tune2fs: Filesystem has unsupported read-only feature(s) while trying to open /dev/loop1
Couldn't find valid filesystem superblock.

I'd have to use a newer OS to do that.

Which version of Kubuntu do you use?

XECDesign, 2018-08-14:
The machine I build final releases on is still running Ubuntu 16.04.3 LTS.

procount, 2018-08-14:
I've installed "Debian with Raspberry Pi Desktop" (RPD86).

- No ADFS/Cumana problems.
- Kali can now be processed.

But building PINN doesn't get very far.... it fails on host-gmp

```
checking size of mp_limb_t... 4
configure: error: Oops, mp_limb_t is 32 bits, but the assembler code
in this configuration expects 64 bits.
You appear to have set $CFLAGS, perhaps you also need to tell GMP the
intended ABI, see "ABI and ISA" in the manual.
package/pkg-generic.mk:172: recipe for target '/home/pi/noobs_test/buildroot/output/build/host-gmp-6.0.0a/.stamp_configured' failed
```

I think configure is getting in a twist as RPD is 32 bits, but because I'm running in a VM on a 64 bit machine, it wants to use ABI=64. How/where do I force ABI=32 ?
(Using gcc 6.3.0)

maxnet, 2018-08-14:
>Which version of Kubuntu do you use?

18.04
Has been upgraded several times from older versions though, so may be running it with different set of packages than you get when you run a fresh install.

>I think configure is getting in a twist as RPD is 32 bits, but because I'm running in a VM on a 64 bit 
>machine, it wants to use ABI=64.

I don't think any distribution other than RPD uses the 64-bit kernel on 32-bit userland stuff, and no development tool will be expecting to run under such environment...
Either install a 32-bit kernel (`apt-get install linux-image-686 && apt-get remove linux*amd64`), or get a different distribution...

procount, 2018-08-14:
Ah! I didn't realise it was running in that mode. 
If only I had thought to do a `uname -a`.
Thanks.

procount, 2018-08-14:
That's fixed it. Now compiling away quite happily. And my first (purely subjective) opinion is that it seems to be running faster - less "thunking" I suppose.

procount, 2018-10-04:
Result = use "Debian with Raspberry Pi Desktop" (RPD86).

closed 2018-10-04 by procount