raspberrypi/noobs #516: Can no longer install anything from NOOBS

Jamesking56, 2018-08-26:
Hello,

I have a Raspberry Pi 2 Model B with a pre-built NOOBS micro-sd card running NOOBS v1.3.12

I want to reinstall Raspbian Lite, however when I try to install I get an error:

```
Error downloading or extracting tarball
tar: can't make dir ./lib/modules/4.14.50+/kernel/drivers/pps/clients: Input/output error
...
```

I thought it might be a NOOBS bug, but I can't update NOOBS since its a pre-built card purchased with the Pi.

So what can I do? I tried Raspbian and other OSs but all have errors on install.

procount, 2018-08-27:
You don't mention what size your SD card is.
If you are connected to the internet, NOOBS will pick the latest version of Raspbian which nowadays needs 8GB or more, whereas the older version (circa NOOBS 1.3.12) only needed 4GB or so. Maybe that is the problem?
Maybe just get another SD card (I would recommend 16GB these days) and install the latest version of NOOBS on it.
There's nothing to stop you upgrading the version of NOOBS on your existing SD card, but if it is only small (< 8GB) I wouldn't bother, because few of the recent OS versions would fit on it.

Jamesking56, 2018-08-28:
@procount I think my card is 8GB which probably means you are right. However, I thought the Raspbian Lite install was much smaller?

procount, 2018-08-28:
Ah, I missed you were trying to install the Lite version, which is pretty small, so space is probably not the issue. Maybe a failing SD card instead. In any case, NOOBS 1.3.12 is pretty old and I would suggest you use the latest version instead. 
Why not take the opportunity to wipe the card by using SDCardFormatter and give it a good test using h2testw (or similar). Unzip the latest NOOBS-Lite onto it (instead of NOOBS-full to save space) and from there you can just install Raspbian-Lite from the internet.

Jamesking56, 2018-08-28:
OK, how do I format the card? When I tried with gparted it said it was formatted but really it wasn't, I plugged the card back into the Pi and everything was still there.

I'll give SDCardFormatter a go but I have a feeling that it won't work because I bought the card pre-built with NOOBS, so it probably has some sort of write protection on it for certain partitions.

procount, 2018-08-28:
It will be a standard SD card, but it will be partitioned. It will not be write protected. See the following section of the NOOBS README file, and reading the rest of it will probably help you get the most out of NOOBS.

https://github.com/raspberrypi/noobs#how-to-format-an-sd-card-as-fat

The link to SD card formatter will now download v5 which does not have the "Format size adjustment" option anymore as it does this by default.

Jamesking56, 2018-08-28:
Yep, tried another card and managed to reformat it fine. Closing as I think this is a hardware issue, thanks for all your help :+1:

closed 2018-08-28 by Jamesking56