raspberrypi/noobs #517: Screenly Update

procount, 2018-08-29:
os_list_v3.json needs a small update to allow Screenly to install on the model 3B+, which is now supported since v0.15 (https://github.com/Screenly/screenly-ose/releases/tag/v0.15).

See also https://www.raspberrypi.org/forums/viewtopic.php?f=56&t=221548)

XECDesign, 2018-08-29:
I'd need a ping from @vpetersson

vpetersson, 2018-08-31:
@XECDesign - I'm traveling this week - will hopefully have time to look at it next week.

vpetersson, 2018-09-12:
Sorry - running a bit behind on this but it's blocked by [this issue](https://github.com/Screenly/screenly-ose/issues/711).

vpetersson, 2018-09-12:
In short - we will add support in the next version.

vpetersson, 2018-11-23:
@XECDesign - Sorry for the slow turn-around here, but I've now pushed an update that includes B+ support. (We've also adopted pi-gen as the build-processes for disk images to simplify this further going forward.)

XECDesign, 2018-11-26:
Thanks, Viktor. It should be back on the list in NOOBS for all Pi 3 models.

closed 2018-11-26 by XECDesign