raspberrypi/noobs #518: Error when installing on HDD: returns negative disk space

psharma04, 2018-10-08:
When installing Raspbian on a 2TB external drive, NOOBS returns the error: Not enough disk space. Need 4405MB, got -189554MB. Possibly an issue with drive exceeding maximum disk size?

procount, 2018-10-08:
See #418

procount, 2018-10-08:
@psharma04  - it would be really helpful if you could perform the same exercise using PINN instead of NOOBS where the #418 fix has already been implemented, to test it works fully before being incorporated into NOOBS. 

See https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&sid=4e355893299cd9f1a318546c820032b5
(The link to download pinn-lite.zip is near the end of that first long post.)

XECDesign, 2018-10-08:
A test build of NOOBS with the fix:
https://drive.google.com/file/d/19ilqrxu7K3--WjHZno3jNOpRRbWFaupE/view?usp=sharing

XECDesign, 2018-10-16:
The fix should be in the latest release of NOOBS.

Closing due to lack of response.

closed 2018-10-16 by XECDesign