raspberrypi/noobs #520: Recallbox doesn't boot on RPi3

frogforge, 2018-11-11:
NOOBS_v2_9_0 on RPi3

After installing Recallbox via NOOBS and selecting it from boot screen, Recallbox never boots. Instead it hangs on a black screen with monitor/screen indicating no signal.

Raspberian and libreELEC installed at the same time boot and operate fine.

Installing the Recallbox image from recalbox.com onto the same sdcard (recallbox.img.xz with Etcher) boots and operates fine.

XECDesign, 2018-11-11:
Are you able to check whether it actually fails to boot or if it fails to output anything to the screen? Does it show up on the network, for example?

ping @digitallumberjack

procount, 2018-11-11:
I tried recalbox with PINN on my 3B (same kernel/firmware as NOOBS 2.9 suitable for 3B+) and it did the same thing - just a black screen and no error codes on the LEDs.
Then I downgraded PINN's firmware (in the menu) to make it compatible with old OSes, and then recalbox worked ok.
So I think recalbox is not suitable for use with the latest "3B+" firmware and needs upgrading.

Now I've only just upgraded PINN's firmware to match NOOBS 2.9, so I don't know if it's the latest firmware update that has caused the problem, or if it was the original 3B+ firmware (needs more investigation).

Nevertheless, you can get recalbox to work in PINN by ensuring PINN uses the old firmware (which it will automatically if you do a fresh install on a 3B)

frogforge, 2018-11-11:
> Are you able to check whether it actually fails to boot or if it fails to output anything to the screen? Does it show up on the network, for example?
> 
> ping @digitalLumberjack

Good point.

To test, I turned on the Pi and started pinging it. That resulted in request timeouts as expected. But about 15 seconds after selecting Recallbox from boot menu, the pings started succeeding. They again timeout when I turned off the Pi.

So it does seem to be no output, not failure to boot.

XECDesign, 2018-11-11:
Ah, then it's a known issue. Might be worth taking it off the list for the 3B+ in NOOBS in the meanwhile.

procount, 2018-11-12:
Is the issue with the latest firmware/kernel in NOOBS, or with recalbox itself? It seems odd it doesn't affect the other distros (AFAIAA).

XECDesign, 2018-11-12:
They're shipping very old firmware "Apr  4 2017".

XECDesign, 2018-11-12:
Wait, why would it show up in NOOBS 2.9? I see "Pi 3 Model B Rev" in `supported_models` in `os_list_v3.json`, which doesn't match the string for 3B+.

procount, 2018-11-12:
The OP said he is using the 3B, NOT the 3B+, but NOOBS still uses the 3B+ compatible firmware, which is not compatible with the VERY OLD recalbox firmware.

XECDesign, 2018-11-12:
Sorry, misread your post when skimming through it again.

Then maybe it should be taken off the list altogether. The issue has been around for a long time.

XECDesign, 2018-11-12:
> I thought you meant that booting with a black HDMI screen was a known issue, separate from the mismatched firmware issue.

It's the same issue. There's no HDMI output because of start.elf in recalbox doesn't support bootcode.bin in NOOBS.

procount, 2018-11-12:
I've copied the Recalbox entries to PINN's "os_list.json", so feel free to delete it from NOOBS as you wish.