raspberrypi/noobs #521: V3?

procount, 2018-11-16:
V3.0.0 implies a new NOOBS version on top of the new version of bundled Raspbian.
There only seems to be 1 tiny commit on the repo since 2.9, so is there anything else that has been added?
(Sorry for being impatient :wink: )

XECDesign, 2018-11-16:
The only uncommitted change that I'm aware of is changing the recommended image from Raspbian to Raspbian Full. Nothing important.

procount, 2018-11-16:
Cool!
I still only see Raspbian Lite and Raspbian in the os_list_v3.json file, though. Is Raspbian Full missing (or did I just miss it)?

XECDesign, 2018-11-16:
> Is Raspbian Full missing (or did I just miss it)?

It was missing. Good catch, thanks.

> EDIT: I can't download ATM, but I assume NOOBS full contains Raspbian Full (not Raspbian)?
Yup

closed 2018-11-16 by procount

XECDesign, 2018-11-16:
Heads up I had to do a forced push just then, because the previous patch was not applied to the internal repo when NOOBS was built.

procount, 2018-11-16:
Cheers.