raspberrypi/noobs #527: Non-translated strings

lurch, 2019-01-09:
The QMessageBox [here](https://github.com/raspberrypi/noobs/blob/master/recovery/main.cpp#L246) doesn't use the `tr` stuff, and so will always be displayed in English, even if NOOBS is set to a different language.

maxnet, 2019-01-09:
That's an internal error that should never occur though.

The fact that it sometimes does is probably a bug here: https://github.com/raspberrypi/noobs/blob/master/recovery/main.cpp#L85
Probably needs a `&& !devname.endsWith("0")` to catch partitionless SD cards.

procount, 2019-01-10:
What about partitionless USB sticks? Will they need an extra condition?
(I imagine some users will be able to create such things!)

lurch, 2019-01-10:
Ah, I only skimmed the code quickly :wink: