raspberrypi/noobs #528: Can't exit gracefully with noobs lite

FrankvdStam, 2019-01-15:
Using a raspberry pi model 3B.

Downloaded:
NOOBS Lite
Network install only
Version: 3.0
Release date: 2018-11-16

Plugged in an Ethernet cable, got a green and orange light going.
The GUI then asks for a wifi network, I only have the cable. Not sure
if this is on purpose? Why can't I use the cable to download an OS?

Guess I'll have to get the pre-loaded version of noobs, which is fine.
However I can only select wifi networks in the GUI. All other options 
are greyed out, can't click on exit. I can just yank the cable out but I'd
prefer to shut it down properly.

[image]

procount, 2019-01-15:
NOOBS uses dhcp to get an IP address from your router, so plugging in an ethernet cable is normally all that is required. It works best if you have it connected before you power-on the RPi. The wifi dialog only pops up if you haven't got an ethernet connection, so it looks like your RPi is either not getting an IP address or is not able to reach the internet through your router.

You could drop into the recovery shell (press ctrl-alt-F2) and login (user root, password raspberry) and then type `ifconfig` to find out the status of you network connection. I don't think a preloaded version of NOOBS will help you because from your screenshot it seems to have been installed just fine.

The other options in the menu will be greyed out until you have a network connection, have a list of OSes to install (from internet or USB stick) or have installed some OSes (depending on which button we are talking about).
The EXIT button exits the NOOBS recovery menu to the boot selection dialog, which you can only do if you have installed an OS that you can boot to. 
NOOBS runs from RAM and takes care to keep its partitions read-only except when necessary. So provided NOOBS is not accessing the SD card or USB stick, like when installing an OS, there is no need to perform a safe shutdown before pulling the plug

FrankvdStam, 2019-01-15:
Okay, thanks for your time and explanation. I'm not on my own network, turns out the local sysadmin has set up a whitelist for internet access which explains why I'm not getting much out of the cable.

closed 2019-01-15 by FrankvdStam

lurch, 2019-01-16:
> I don't think a preloaded version of NOOBS will help you because from your screenshot it seems to have been installed just fine.

I assume by "preloaded version" he meant NOOBS Full, which would allow him to install Raspbian even if NOOBS itself isn't able to access the internet?

FrankvdStam, 2019-01-16:
@lurch Yes, your assumption is correct, I meant NOOBS with a pre-loaded OS. I ended up waiting till I got home. I used lite from my home network, which worked fine. 

I didn't want to get the full version because I don't need loads of the programs that come preinstalled on it.