raspberrypi/noobs #530: Help request building kernel driver

procount, 2019-01-30:
QUESTION ONLY
I'm trying to develop a built-in touchscreen driver for PINN, within the NOOBS development environment, but I'm finding the edit-build-debug cycle a bit long winded. Each time, buildroot is building recovery & the 2 kernels, before I download it to the RPi and test it.

I was wondering if anyone knew of a quicker way?
Some things I thought of:

1. Could I include support for modules in the current NOOBS buildroot kernel, then just build my driver as a module and test it in Raspbian (or even in PINN)? It would then be built as a built-in driver for PINN when it is complete.
2. Should I bite the bullet and build the whole kernel for Raspbian and develop it as a module within that environment first?
3. Is there some way to force buildroot to just build my driver files and link it to the rest of the kernel, without rebuilding everything?
4. Any other ideas?

maxnet, 2019-01-30:
RTFM :-)
https://buildroot.org/downloads/manual/manual.html#_infrastructure_for_packages_building_kernel_modules

procount, 2019-01-30:
Ta! Didn't think to look there, as my recollection was that buildroot insisted on building EVERYTHING since it didn't keep track of dependencies, so thought it had to be something special. But I guess modules are special :)

maxnet, 2019-01-30:
Good excuse to upgrade then. ;)

While you are at it, you may also want to look at using buildroot in a way that makes upgrading to newer buildroot versions easier, by having your own package directories external to buildroot.

See scriptexecutor implementation notes for some background how that can look.

https://github.com/raspberrypi/scriptexecutor/wiki/Building-and-customizing
https://github.com/raspberrypi/scriptexecutor/

lurch, 2019-01-30:
> Good excuse to upgrade then.

Last time I did that, it wasn't easy... https://github.com/raspberrypi/noobs/commit/f7e730ab05ee60503af46f7d866c5c956b83743c

maxnet, 2019-01-30:
>Last time I did that, it wasn't easy...

Nobody said things should be easy. :smiley:

Probably easier then last time though.
Also note that with newer kernel branches it is possible to build a single multiarch kernel that can boot on both Pi 1 and later Pi. Not strictly necessary anymore to build two separate ones anymore.

procount, 2019-01-30:
That looks like some interesting food for thought! 
I might not implement such a huge change right at the moment, though. 
I don't think there is any impetus for NOOBS to do this, so it would just be for PINN, and would mark a significant departure away from NOOBS
I'm still wading through how to specify GPIO pins properly in DTS files. :smile:

closed 2019-02-10 by procount