raspberrypi/noobs #532: "Error saving boot files to memory. SD card may be damaged."

Fateofiko, 2019-02-15:
Hi everyone :) !
I changed the recovery/config.h file and i made my own build of NOOBS . The only thing that i have change are this two options:
Original:
#define RESCUE_PARTITION_SIZE  63
#define MAXIMUM_BOOTFILES_SIZE 64  

My setup:
#define RESCUE_PARTITION_SIZE  4499
#define MAXIMUM_BOOTFILES_SIZE 4500

My idea is to make the "recovery partition" to be bigger, and this way i will be able after installation to change the recovery image with another one. And then i can put "runinstaller" option to recovery.cmdline file and i can reinstall everything! And this is very useful for me, because i can change the image remotely and reinstall everything remotely!

The problem is, when i copy everything to the sd card including the image files to the "os" folder, and when i start the raspberry, the process stops with this error :
"Error saving boot files to memory. SD card may be damaged."
If i do not put the image files to the "os" folder, everything pass successful !
May be the problem is that, when i copy all the files to the SD card, the "cp" command leave this warning to the output of the terminal:
"cp failed to preserve ownership for operation not permitted "
I did a lot of stuff to preserve the ownership but nothing helps!!!

Can you tell me how to make it right, i just want the "recovery partition" to be fixed size (let say 4GB)?

lurch, 2019-02-15:
That error message is coming from https://github.com/raspberrypi/noobs/blob/master/recovery/initdrivethread.cpp#L345 which is failing because the RPi doesn't have enough RAM to (temporarily) store several gigabytes of data while the SD card is being reformatted.
So instead of the test [here](https://github.com/raspberrypi/noobs/blob/master/recovery/initdrivethread.cpp#L42) causing `method_reformatDrive()` to get run, you want to put `MAXIMUM_BOOTFILES_SIZE` back to normal so that `method_resizePartitions` gets run.

From the description of your problem, it sounds like what you _actually_ want to do is modify the `newSizeOfRescuePartition` value [here](https://github.com/raspberrypi/noobs/blob/master/recovery/initdrivethread.cpp#L145) (e.g. by changing the number 100 at the end).

And don't forget that NOOBS works with tarballs and not "images" :wink: 

The permissions message is harmless (and can be ignored), it's simply because FAT32 doesn't support the same file-permissions that most Linux filesystems use.

procount, 2019-02-15:
It's a useful feature. Which is why you will find it in PINN already, but it is parameterised there instead of being fixed.

You don't need to use the runinstall option each time you update the os folder. Just install your OSes again. (With PINN you can just update or re-install single OSes without having to do all of them together)

The runinstall option re-partitions the whole card which is only necessary when first installed. During this time it tries to copy all of the recovery partition to memory whilst it does this. IIRC it does this if the partition size is small and does something slightly different if larger.

Lack of ownership is probably because you are using linux to copy files to a FAT partition that does not support ownership. You can probably safely ignore that warning.

You could sidestep this issue by putting your OSes on a usb stick instead.

lurch, 2019-02-15:
Ahhh sorry @procount , I forgot to say to @Fateofiko that if you're doing non-standard stuff, you're probably much better off looking at [PINN](https://github.com/procount/pinn) than you are [NOOBS](https://github.com/raspberrypi/noobs) :smiley:

procount, 2019-02-16:
That's quite alright @lurch. You added the detail I couldn't at the time. :)

Fateofiko, 2019-02-17:
Thank you very much, i appreciate your detailed answers! This is really helpful for me!
NOOBS is great project, and  YOU are great team!

Thank you all !!!

Agungsetiawanpawallo, 2019-03-13:
OMG, it's make me stress. i was do all method to install that. i don't know why, when looking first tutorial in youtube it looking so easy, but when did like that error, failed, cannot find the drive with NOOBS file was came. i was tried to install that from 1 week ago till now and never success, i have another problem right now. 
would you help me please ?

 I'm a student from Indonesia and i want to use Raspberry Pi for my research.
thank you

Agungsetiawanpawallo, 2019-03-13:
I'm used PINN for install that but something wrong. nothing NOOBS Raspbian showing over there.

Agungsetiawanpawallo, 2019-03-13:
I'm used PINN for install that but something wrong. nothing NOOBS Raspbian showed over there

JamesH65, 2019-03-13:
If you just want to install an OS, the best place to ask is the forums, lots of information there. Try the official docs first. https://www.raspberrypi.org/documentation/installation/installing-images/README.md

Agungsetiawanpawallo, 2019-03-13:
First my Raspberry pi is good, but when i want reinstall to add OpenCPN software, it can not install again, and tried some method that i was read from blog and also youtube, then more problem came. 

thanks before for help me to solve my problem.

Agungsetiawanpawallo, 2019-03-13:
Thank you Mr. James

what do think good NOOBS or Raspbian?

Regard Agung Setiawan Pawallo