raspberrypi/noobs #533: Q:building DTBO files

procount, 2019-02-27:
Noobs' buildroot does not build dts files but pulls them in from rpi-firmware. (Why?)
I have a couple of dts files I need to build. What is the easiest way to incorporate them into buildroot?
Should I create another package?
Should I enable building of all DTSs and just add them to that folder?
And I suppose the DTC supplied in buildroot is not the correct one for the raspbian overlay/dtparam mods?

XECDesign, 2019-02-27:
> Noobs' buildroot does not build dts files but pulls them in from rpi-firmware. (Why?)

IIRC, there was a time when the kernel and firmware were pulled from rpi-firmware, so it may be reusing that approach because it was quicker than doing it properly. Pretty sure there was another reason, but it was too long ago to remember.

Whatever it was, it's not relevant now and the right thing to do would be to build them along with the kernel.

> I have a couple of dts files I need to build. What is the easiest way to incorporate them into buildroot?
Should I create another package?
Should I enable building of all DTSs and just add them to that folder?

Are the dts files something that should be merged into the linux repo anyway? Then they'd just end up in the overlays directory when they're pulled from rpi-firmware.

If not, it's really up to you. I'd either add it as a kernel patch, or add a new package. The latter would probably keep things cleaner.

> And I suppose the DTC supplied in buildroot is not the correct one for the raspbian overlay/dtparam mods?

If it's recent enough, it should be fine, since the necessary changed are upstream. I found the very latest version is a bit buggy, but the previous release should be fine. If the current is too old, I suspect you can just bump the version and it will be fine.

procount, 2019-02-27:
Ok. I have no idea how to build them properly but willing to give it a go. Any specific config options you know of that I should set? Otherwise I'll just muddle my way through.

Ha! Not sure the associated drivers are worthy of being upstreamed as I broke all the rules. I got fed up of switching config files around for different displays (hdmi, rpf touchscreen, hyperpixel 3.5" & 4.0") so I wrote combined kernel drivers for the hyperpixel touchscreens AND their LCD init commands. So now PINN can detect which display is fitted and can configure itself appropriately so I always get a display no matter what I fit (with a user preference if more than 1 is fitted). It all works nicely, but I'm just trying to figure out how to integrate it into the proper build process.

I suppose the hyperpixel 3.5 touchscreen driver could be upstreamed to replace their python driver but I think they've discontinued it now the 4.0 exists. 

dtc seems to be this one....

#define DTC_VERSION "DTC 1.4.4-g756ffc4f" from linux-12d78096b1

XECDesign, 2019-02-28:
That version of dtc should be good.

No specific flags. If you're making it a separate package, the typical command to build overlays should be fine. 
https://github.com/raspberrypi/documentation/blob/master/configuration/device-tree.md#21-fragments

If you're patching the kernel and making all overlays build as a part of the kernel, then you just need to make sure it's in arch/arm/boot/dts/overlays and in the makefile (don't remember if there's a kconfig file as well, but probably not). Then make sure they get copied across and that should be about it.

procount, 2019-02-28:
Maybe I'm looking with my elbows, but I can't find anything in linux.mk that suggests it would compile the overlays. There are mentions of compiling the main dtb to be appended to the image, but nothing useful about overlays. And even a `make dtbs` from the buildroot folder isn't doing anything as it can't find a rule for that.

I tried following this article -> https://jumpnowtek.com/rpi/Compiling-Raspberry-Pi-Overlays-with-Buildroot.html but that didn't compile the overlays either (It seems to be for a later version of buildroot, but even attempting to modify it accordingly didn't seem to work). Maybe it's missing Kconfig and Makefile for the DTBOs?

I'm beginning to think creating a package to compile my 2 DTS files and copy them to the overlays folder would be a much easier way forward.

XECDesign, 2019-02-28:
Are you sure we're not using device tree files built with the kernel?

https://github.com/raspberrypi/noobs/commit/e385dfaa3d589204d27c3160ad654d48846ee308#diff-03658f2fadbdfb904bf63f8af9ccc6d2

procount, 2019-02-28:
_NOTE: When I say I "broke all the rules" concerning my drivers, I mean that I merged the touchscreen driver with the LCD init code to create a single driver for the HAT, rather than individual drivers for each chip. This is because I am using the detection of the touchscreen as an indicator that the whole HAT is present, as it is not possible to detect the LCD on it's own. So I guess it would be frowned upon upstream(?). But I suppose it's not much different to a driver that loads firmware into a device, it's just at the HAT level not at chip level. I haven't yet checked them against the checkpatch.pl script either (which I believe would be a pre-requisite). The drivers can even be used to detect the HAT without installing themselves so I can switch to the appropriate one._

Since I've now migrated the Pimoroni userland drivers to pure kernel drivers (including the full 4 rotation possibilities), they don't need any extra support functions from PINN (just an appropriate config.txt file) so I believe they could be easily added to NOOBS too. Let me know if that would be of interest and I'll submit a patch later. (I know you're not including any new features now, but just thought I'd throw it out there :smile: )

Maybe when I've finished and tidied it all up, you could provide an assessment of whether you think upstreaming for inclusion in Raspbian etc. would be possible/useful?

procount, 2019-02-28:
Oh. F*S. Did I just forget to add my 2 little files to the Makefile !? I'll try again. Thanks.

XECDesign, 2019-02-28:
I think we'll pass on the patch for NOOBS, since it would need to be maintained for new kernel versions. Since I don't own the hardware, it's not something I can support. The best way would be to get it into the Raspbian kernel and it will naturally end up in NOOBS.

However, I don't know if it would be accepted. @pelwell may be able to offer some insight on that point.

procount, 2019-02-28:
Yes, I thought you'd pass on that ;)

pelwell has already been very helpful with my overlays. https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=232590

And now I have added my files to Makefile, they compile nicely and end up in the overlays folder.
Many thanks.

closed 2019-03-11 by procount